facebookresearch/habitat-sim #1575: Issues with downloading the dataset

sparsh-umd, 2021-11-28:
## Habitat-Sim version
0.2.1

Hi!

I've created a new conda env with
`conda create -n habitat python=3.8.10 cmake=3.14.0`
& installed habitat with
`conda install habitat-sim withbullet -c conda-forge -c aihabitat`

Then I cloned this repo & did `cd habitat-sim`.

Then I tried to download dataset by running the command `python -m habitat_sim.utils.datasets_download --uids habitat_example_objects --data-path dat`. And, it failed with the below output:

Traceback (most recent call last):
  File "/home/username/anaconda3/envs/habitat/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/home/username/anaconda3/envs/habitat/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/username/habitat/habitat-sim/habitat_sim/__init__.py", line 13, in <module>
    import habitat_sim._ext.habitat_sim_bindings
ImportError: generic_type: type "SceneNode" referenced unknown base type "Magnum::SceneGraph::Object<Magnum::SceneGraph::BasicTranslationRotationScalingTransformation3D<float> >"

However, running the same command from outside the cloned directory, downloads the dataset with no issue.

Thanks for the help!

aclegg3, 2021-11-29:
Sounds like it could be a module collision. The system may be picking up local versions of the packages installed on the machine instead of the conda installed packages. By chance is the habitat-sim directory on your `PYTHONPATH` and have you installed a previous version of Magnum or Habitat from source? If so, you may need to clear the `PYTHONPATH` and/or remove the stale build directory to correctly access the module from within the directory.

sparsh-umd, 2021-12-14:
Yeah, I have installed habitat from source previously. I removed it now. No issues anymore.

Thanks!

closed 2021-12-14 by sparsh-umd