facebookresearch/habitat-sim #1576: Issues with loading a custom object with URDF file

sparsh-umd, 2021-11-28:
## Habitat-Sim version
0.2.1

I installed habitat-sim with conda & launched habitat-viewer with physics enabled & tried to add a custom object to the scene by pressing 't' button.

A prompt came in the terminal to provide a urdf path 
`Load URDF: provide a URDF filepath.`
I entered the path
`data/cube.urdf` & hit return.

The simulator exited with the below message:
`E - failed to parse link. Aborting parse/load for  data/cube.urdf`
`Failed to parse URDF:  data/cube.urdf , aborting.`
`ESP_CHECK failed: failed to parse/load URDF file  data/cube.urdf`
`Aborted (core dumped)`

NOTE:
Launching habitat-viewer without enabling physics & trying to import URDF file returns the below error:
`addArticulatedObjectFromURDF not implemented in base PhysicsManager.`
`E1127 18:57:15.689186 11600 ManagedContainerBase.h:203] ::getObjectHandleByID : Unknown ArticulatedObject managed object ID:-1. Aborting`
`E1127 18:57:15.689218 11600 ManagedContainerBase.h:331] <ArticulatedObject>::getObjectCopyByID : Unknown ArticulatedObject managed object handle :. Aborting`
`Segmentation fault (core dumped)`

sparsh-umd, 2021-11-28:
If it helps, [This](https://github.com/bulletphysics/bullet3/blob/master/data/cube.urdf) was the URDF file I used

aclegg3, 2021-11-29:
Hey @sparsh-umd,

You may want to try [enabling a different logging level](https://aihabitat.org/docs/habitat-sim/logging.html) to get more detailed messages regarding the URDF parsing state. E.g. `export HABITAT_SIM_LOG=veryverbose`

On my end the `data/test_assets/urdf/cube.obj` did not exist. If I create the asset, then the URDF loads fine.

closed 2021-11-29 by aclegg3

sparsh-umd, 2021-12-14:
Thanks for the reply @aclegg3!

I tried setting `export HABITAT_SIM_LOG=veryverbose`, launched habitat-viewer & tried to import a URDF file. Still, the terminal output was the exact same as before:

`E - failed to parse link. Aborting parse/load for data/cube.urdf`
`Failed to parse URDF: data/cube.urdf , aborting.`
`ESP_CHECK failed: failed to parse/load URDF file data/cube.urdf`
`Aborted (core dumped)`

Tried to import one of the urdf objects present at `data/test_assets/urdf/`. They were imported fine.

aclegg3, 2021-12-14:
The URDF file is fine but there my be a missing asset if you copied the urdf file somewhere and not the assets it references. If I load that file directly from `src/deps/bullet3/data/cube/urdf` it works fine for me.

sparsh-umd, 2021-12-31:
Yeah, that was the issue. Thanks for the help!