facebookresearch/habitat-sim #1580: support MuJoCo

hpf9017, 2021-12-01:
## 🚀 Feature

support MuJoCo for physical engine
## Motivation

since MuJoCo is going to be open source， is there any plan to support it？

## Pitch

## Alternatives

## Additional context

aclegg3, 2021-12-01:
Hey @hpf9017, thanks for the suggestion. We've discussed this possibility as well. Currently there are no concrete plans for us to develop a Mujoco backend in the immediate future, however we'd be open to collaboration on the topic if someone else was interested in doing so. We may revisit this topic in the future if provided adequate motivation to invest in the work.

As the simulation API is generic and abstracted from the Bullet implementation, this would primarily involve implementing a "MujocoPhysicsSimulator" backend and replicating the functionality in the `src/esp/physics/bullet/` directory with a Mujoco wrapper in a new subdirectory `src/esp/physics/mujoco/`.

aseembits93, 2022-05-06:
@aclegg3 Hope all's well with you. My research is about perception for legged robots and I have some bandwidth over the summer to make Mujoco work with Habitat, guidance and support would be greatly appreciated, Thanks!

aclegg3, 2022-05-06:
Hey, @aseembits93. Sounds good, feel free to DM me on our [Slack workspace](https://aihabitat.org/#contact) when you have questions. 

Roughly I'd suggest starting by forking habitat-sim, adding Mujoco codebase as a submodule, setting up the buildsystem to support installation with --mujoco, and testing with a MujocoPhysicsManager (see [BulletPhysicsManager](https://aihabitat.org/docs/habitat-sim/classesp_1_1physics_1_1BulletPhysicsManager.html)). You can stub out the methods and test just initializing the simulator object first. Add an option for "mujoco" to the [physics config file](https://github.com/facebookresearch/habitat-sim/blob/ec81c8cfe50b35f98edc4a390ee9b46a961b7144/data/default.physics_config.json#L2) to trigger this option.

aseembits93, 2022-06-30:
Hi @aclegg3 , the Slack link you provided no longer works.

aseembits93, 2022-06-30:
> Hi @aclegg3 , the Slack link you provided no longer works.

@Skylion007 lmk if you could me to the slack workspace, have a good day everyone.