facebookresearch/habitat-sim #1582: About /habitat/examples/viewer.py

posshe, 2021-12-02:
## Habitat-Sim version
v0.2.1
Hello,
I have some problems about the simulator. I want to control the robot by hand according the simultaneous feedback from the viewer. I try to modify the viewer.py to realize the function. But It can not work and report the "Segmentation fault" during the line "self.sim = habitat_sim.Simulator(self.cfg) ". And I thought this problem is similar with (https://github.com/facebookresearch/habitat-sim/pull/1056).
And I install the habitat-sim using the command 
> conda install habitat-sim withbullet -c conda-forge -c aihabitat
Is it the version error between viewer.py and simulator.cpp?
Look forward for your reply.
Thank you

posshe, 2021-12-02:
I find another version about viewer.py, and other error occurs. `RenderTarget::Impl::blitRgbaToDefault(): the viewport size does not match between the internal frame buffer and the default frame buffer` . If I comment the line `self.render_camera.render_target.blit_rgba_to_default()`, the viewer could run but the visualization is all black and nothing shows.
Thank you for your help.

aclegg3, 2021-12-02:
Hey @posshe,

Conda build from the previous release version won't support the viewer.py application. I'm not sure if the conda nightly would work, as we haven't tested it, but I am doubtful. 

I suggest you install from source on the `main` branch instead and try again. Let me know on this thread if that works for you or if other issues emerge.

posshe, 2021-12-03:
Hello @aclegg3 ,
Thank you for your nice reply. I will try it by installing from source. If some other issue occurs, I'll tell you. Thank you again~

closed 2021-12-06 by Skylion007