facebookresearch/habitat-sim #1583: Robot intertia while navigating a robot agent

ruddradev, 2021-12-02:
## Habitat-Sim version
v0.2

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/ 

Yes, I have looked at the docs and issues, could not find out a concrete solution.

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Yes, I looked at the nav tutorials.

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Hello, I was wondering if the robot Inertia is taken into account in some way while moving in 2d using a robot agent. When I ran the tutorials with Habitat 0.2(The current release which I installed with Bullet), it appears that the robot inertia does not come into play. When I take a left turn and then immediately take a right turn, the robot should turn less to the right. As I had experienced in iGibson.

Can someone please clarify if I am doing something wrong, or this is a feature of a later release?

aclegg3, 2021-12-02:
Hey @ruddradev, your observation about the tutorial code is correct. Those are using direct velocity or positional control. 

However, if you want more accurate dynamics you could write an acceleration or torque based controller for 2D motion on the navmesh. Simply integrate velocity from a higher order control signal. You could also use fully dynamic simulation with Bullet and actuate the wheels or legs of the robot directly.

ruddradev, 2021-12-03:
Hi @aclegg3 : Thanks for the clarification. How do I do a fully dynamic simulation using bullet? Is there an API inside Habitat to enable that? Or do I need to do the calculations outside the Simulator and actuate based on the velocities through the velocity control?

aclegg3, 2021-12-03:
Either option is feasible. 

If you are able to import a robot URDF then you already have the [full API](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedBulletArticulatedObject.html) enabled. If your URDF defines wheel or leg links/joints you can directly actuate those degrees of freedom with position/velocity/torque control or setup JointMotors to do PD control. [This test ](https://github.com/facebookresearch/habitat-sim/blob/19f9cb7f746257762c5de3433332fd89bfd688ee/tests/test_physics.py#L1016)and others around it provide some examples of using the ArticulatedObject dynamics API for reference.

If you are not interested in dealing with the dynamics of wheeled motion or legged locomotion and would prefer a 2nd order abstraction of base position control on the NavMesh, you can instead integrate accelerations or forces manually and then apply the resulting velocities. The RigidBody [VelocityControl](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.VelocityControl.html) structure is really just [integrating](https://github.com/facebookresearch/habitat-sim/blob/19f9cb7f746257762c5de3433332fd89bfd688ee/src/esp/physics/RigidObject.cpp#L100) velocity manually to update 6DoF state, so you could follow this pattern, adding one more step to integrate acceleration or torque first.

ruddradev, 2021-12-06:
Hi @aclegg3 : Many thanks for the pointers. I will look into them.

closed 2021-12-06 by ruddradev