facebookresearch/habitat-sim #1587: About the torque feedback between the robot's joint

posshe, 2021-12-07:
## Habitat-Sim version
v0.2.1
Installed by source.

Hello,
There's some questions about the force/torque feedback in the habitat. The urdf of the robot has been imported into the habitat, and I want to control the robot by the torque controller. But I can't get the force/torque feedback between the joins. I try to use the ` joint_forces: typing.List[float] get set
    Get or set the joint forces/torques (indexed by DoF id) currently acting on this Articulated Object.` (in https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedBulletArticulatedObject.html), but the result that the function returns are all zero. Is there any function to get the force/torque? 
Thank you very much!
Bless~

aclegg3, 2021-12-07:
Hey @posshe,

Your observation is correct. The `joint_forces` property only gets/sets externally applied torques by the user and is cleared after each simulation step.
PR #1585 adds a function to query the torques applied at the most recent step by the JointMotor (PD) controllers. This may be more useful to you.

posshe, 2021-12-08:
Hello, @aclegg3 
Thank you for your nice reply. It's really helpful for my work. I will close the issue.
Thank you again.
Bless

closed 2021-12-08 by posshe