facebookresearch/habitat-sim #1600: Could we build a costumed scene in habitat for 3d reconstruction for an object

wwtx9, 2021-12-17:
## ❓ Questions and Help
Is there a scene for 3d reconstruction of an object. For example, there is only one object in an empty ground in habitat?

aclegg3, 2021-12-17:
Hey, @wwtx9. It sounds like you want to be able to render synthetic views of a single imported 3D asset (e.g. a coffee mug model) sitting at rest on a box or ground plane.

If this is an accurate interpretation I'd suggest instantiating and instance of the Habitat simulator with no assets and then adding a STATIC primitive box object to act as the ground followed by your object of interest. An example can be found in [this unit test](https://github.com/facebookresearch/habitat-sim/blob/69ae4848503d5dcc74d6b5920957c1a641ef0a9b/tests/test_robot_wrapper.py#L27). Just import an object on the ground instead of a robot.

wwtx9, 2021-12-20:
Hi, @aclegg3 thank you for taking your time to reply to my question. 

My ideal scene is a horizontal ground with an object. And then set a robot with a stereo camera which can move around with the object. 

I attached a link [a vehicle on the ground](https://youtu.be/of4ANH24LP4?t=65) to help me clarify what is the scene that I want looks like.

I look forward to hearing back, thank you!

aclegg3, 2022-01-04:
Hey @wwtx9, thanks for the example. This sounds feasible to me using existing APIs in Habitat. I'd suggest trying out the [tutorials](https://aihabitat.org/tutorial/2020/) to understand system and sensor setup and taking a look at the unit test I linked for an example of setting up the ground plane and rendering images.

closed 2022-01-04 by aclegg3