facebookresearch/habitat-sim #1602: How to use semantic data for Gibson

kamalesh0406, 2021-12-17:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/
Yes
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
Yes
Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

I generated the .ply, .id, .scn files for the gibson dataset using the code given in tools/ . After generating this data, how should I use them to get the data like level.id, region.id as mentioned in this tutorial [https://aihabitat.org/docs/habitat-api/habitat-api-demo.html#scene-semantic-annotations]().  I am getting an empty list when I print scene.levels.

erikwijmans, 2021-12-18:
This is expected. The Gibson dataset doesn't have region annotations.

closed 2021-12-18 by erikwijmans

kamalesh0406, 2021-12-18:
Okay, but the gibson dataset has room based annotations like room_id etc, is it possible to fetch this data from the simulator or the agent?

sky-story, 2024-05-12:
> Okay, but the gibson dataset has room based annotations like room_id etc, is it possible to fetch this data from the simulator or the agent?

@kamalesh0406  Did you fetch the room_id from iGibson successfully？Could you please share the method or code？I will very appreciate it！