facebookresearch/habitat-sim #1603: Agent stuck when reproducing the "Embodied Continuous Navigation" tutorial.

Goooyi, 2021-12-22:
## Habitat-Sim version
v0.2.1

## Docs and Tutorials
Did you read the docs?  Yes

Did you check out the tutorials? Yes
## ❓ Questions and Help
I have read the [Habitat-sim Interactivity](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Interactivity.ipynb) and I want to reproduce the "Embodied Continuous Navigation", what I do is I load a mp3d scene(1LXtFkjw3qL) and copy the code in the google colab, also load the `locobot_template_handle` as my agent, find navigable points in the nevmesh and then defined 2 path, sometimes the agent can finished the path successfully but more often it will stuck during the 2. path, I checked related functions especially the `track_waypoint()` function but couldn't find where the problems, the way the agent stuck is that it jaggle around between two angels, say I set `angular_error_threshold = 0.5` in the `track_waypoint()` function, then it stucked and the `angular_error` will only jump between 0.543107 and 0.542775 and not decrease as the `track_waypoint()` should have done. Can you give me some advice? Thanks!

Also I found in the code that, after using `vel_control.integrate_transform()` to get the `target_rigid_state`, instead set the translation of the agent to `target_rigid_state.tranlation`, `sim.step_filter()` is used to get a `end_pos` to place the agent, can you explain why you do that? I set the agent location directly using `target_rigid_state.tranlation` and get the agent successfully make the two paths, I wonder what's the difference? 

this is the path setting part of my code:
```
while not found_path:

    origin = sim.get_translation(locobot_id)
    path1.requested_start = origin
    path1.requested_end = mn.Vector3(sim.pathfinder.get_random_navigable_point())
    path2.requested_start = mn.Vector3(path1.requested_end)
    path2.requested_end = mn.Vector3(sim.pathfinder.get_random_navigable_point())

    found_path = sim.pathfinder.find_path(path1) and sim.pathfinder.find_path(path2)

if not found_path:
    print("Could not find path to object, aborting!")

vis_objs = []

recompute_successful = sim.recompute_navmesh(sim.pathfinder, default_nav_mesh_settings)
if not recompute_successful:
    print("Failed to recompute navmesh 2!")

continuous_path_follower = ContinuousPathFollower(
    sim, path1, sim.get_object_scene_node(locobot_id), waypoint_threshold=0.4
)

# set fps
show_waypoint_indicators = True  # @param {type:"boolean"}
time_step = 1.0 / 60.0

sim.config.sim_cfg.allow_sliding = True

for i in range(2):
    if i == 1:
        continuous_path_follower = ContinuousPathFollower(
            sim, path2, sim.get_object_scene_node(locobot_id), waypoint_threshold=0.1
        )

    if show_waypoint_indicators:
        for vis_obj in vis_objs:
            sim.remove_object(vis_obj)
        vis_objs = setup_path_visualization(sim,continuous_path_follower)

    start_time = sim.get_world_time()
    max_time = 10.0
    previous_pos = sim.get_translation(locobot_id)
    while (
        continuous_path_follower.progress < 1.0
        and sim.get_world_time() - start_time < max_time
    ):
        continuous_path_follower.update_waypoint()
        if show_waypoint_indicators:
            sim.set_translation(continuous_path_follower.waypoint, vis_objs[0])

        if locobot_id < 0:
            print("locobot_id " + str(locobot_id.object_id))
            break

        previous_rigid_state = sim.get_rigid_state(locobot_id)

        # set velocities based on relative waypoint position/direction
        track_waypoint(
            continuous_path_follower.waypoint,
            previous_rigid_state,
            vel_control,
            dt=time_step,
        )

        # manually integrate the rigid state
        target_rigid_state = vel_control.integrate_transform(
            time_step, previous_rigid_state
        )

        # snap rigid state to navmesh and set state to object/agent
        
        end_pos = sim.step_filter(
            previous_rigid_state.translation, target_rigid_state.translation
        )

        sim.set_translation(end_pos, locobot_id)
        sim.set_rotation(target_rigid_state.rotation, locobot_id)

        # Check if a collision occured
        dist_moved_before_filter = (
            target_rigid_state.translation - previous_rigid_state.translation
        ).dot()
        dist_moved_after_filter = (end_pos - previous_rigid_state.translation).dot()

        EPS = 1e-5
        
        collided =  (previous_pos-sim.get_translation(locobot_id)).dot() < EPS

        previous_pos = sim.get_translation(locobot_id)
    
        # run any dynamics simulation
        sim.step_physics(time_step)

        # render observation
        observations.append(sim.get_sensor_observations())
```

aclegg3, 2022-01-04:
Hey @Goooyi, thanks for your thoughts on this tutorial code, hopefully it was useful for you to understand how one might consume Habitat-sim for continuous navigation.

Sorry to hear you are experiencing issues with the example heuristic controller for tracking rotations. This isn't really meant to be a perfect control function, but rather to provide a simple example of how to use the APIs. Ideally you can use this as a stepping stone to invent or train a replacement controller which works better for your use cases!

Regarding the purpose of the `step_filter` function: We use the NavMesh to constrain agent motion without simulating dynamics and 3D contact testing. The step filter validates and, if necessary, truncates the agent's desired linear step against the NavMesh. By skipping this filter, the agent would be allowed to move through walls and obstacles. Following the computed shortest path through the scene as the example is doing will minimize these artifacts, but be aware of the implications here.

Goooyi, 2022-01-06:
Hi @aclegg3 thanks for the explanation! I will sure use this as a start to do more with the agent. And I have a few more questions, one is that, I seen the agent will stuck during simulation whenever a stairs(going down or up) is encountered, my question is, whether the way you control the agent in the tutorial can go up or down in stairs? I haven't be able to recompute a suitable navmesh for the agent to walk, so I wonder it's have something to do with that.  Another question is that, when I set two robots to walk randomly in the scene, if they bump each other, after `step_filter`, they will go though each other or the agents' desired linear step will be truncated?

aclegg3, 2022-01-06:
>  I've seen the agent will stuck during simulation whenever a stairs(going down or up) is encountered, my question is, whether the way you control the agent in the tutorial can go up or down in stairs?

In general stairs are not a problem for the NavMesh. Low ceilings and missing stair geometry in some scenes can, however, cause issues for certain configurations of the [`NavMeshSettings`](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.NavMeshSettings.html) (used to configure the parameters of the NavMesh for recomputation). There are a few parameters which impact this: `agent_height, agent_max_climb, agent_max_slope, filter_low_hanging_obstacles`.

I suggest you take a look at [this Colab tutorial](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb) from ECCV 2020 which has an interactive NavMesh computation cell (start at "Working with the NavMesh") where you can play with the settings and visualize the resulting top-down map. You can also configure the Simulator to render the current NavMesh as a wireframe by setting the [`navmesh_visualization`](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#navmesh_visualization) property to true allowing you to inspect the NavMesh for a scene. The viewer.cpp or examples/viewer.py can be useful for this kind of debugging.

aclegg3, 2022-01-06:
> Another question is that, when I set two robots to walk randomly in the scene, if they bump each other, after step_filter, they will go though each other or the agents' desired linear step will be truncated?

Currently there is no agent collision logic in the step filter if you are setting agent positions kinematically using only the NavMesh as a constraint. You could add this logic with a custom `step_filter` function that checks other agent positions if this is necessary for your work and we'd be happy to review the code if you decided to contribute it back to the project as a PR.

Alternatively, contact between agents could be modeled dynamically if your robots are simulated using forces/torques as control and have attached collision objects instead.

closed 2022-01-26 by dhruvbatra