facebookresearch/habitat-sim #1606: How to add color and scale attributes to the custom object added to the simulator?

yyf17, 2022-01-01:
## Habitat-Sim version
v1.7.0

## ❓ Questions and Help

```
import os
import git
import sys
import math
import os
import random
import sys

import git
import magnum as mn
import numpy as np

import ipdb

import habitat_sim
from habitat_sim.utils import viz_utils as vut

os.environ["IMAGEIO_FFMPEG_EXE"] = "/xxxxx/bin/ffmpeg"

def remove_all_objects(sim):
    for id_ in sim.get_existing_object_ids():
        sim.remove_object(id_)

def place_agent(sim):
    # place our agent in the scene
    agent_state = habitat_sim.AgentState()
    agent_state.position = [-0.15, -0.7, 1.0]
    agent_state.rotation = np.quaternion(-0.83147, 0, 0.55557, 0)
    agent = sim.initialize_agent(0, agent_state)
    return agent.scene_node.transformation_matrix()

# Set an object transform relative to the agent state
def set_object_state_from_agent(
    sim,
    obj,
    offset=np.array([0, 2.0, -1.5]),
    orientation=mn.Quaternion(((0, 0, 0), 1)),
):
    agent_transform = sim.agents[0].scene_node.transformation_matrix()
    ob_translation = agent_transform.transform_point(offset)
    obj.translation = ob_translation
    obj.rotation = orientation

def make_configuration():
    # simulator configuration
    backend_cfg = habitat_sim.SimulatorConfiguration()
    backend_cfg.scene_id = "xxxxxx/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb"
    assert os.path.exists(backend_cfg.scene_id)
    backend_cfg.enable_physics = True

    # sensor configurations
    # Note: all sensors must have the same resolution
    # setup 2 rgb sensors for 1st and 3rd person views
    camera_resolution = [544, 720]
    sensor_specs = []

    rgba_camera_1stperson_spec = habitat_sim.SensorSpec()
    rgba_camera_1stperson_spec.uuid = "rgba_camera_1stperson"
    rgba_camera_1stperson_spec.sensor_type = habitat_sim.SensorType.COLOR
    rgba_camera_1stperson_spec.resolution = camera_resolution
    rgba_camera_1stperson_spec.postition = [0.0, 0.6, 0.0]
    rgba_camera_1stperson_spec.orientation = [0.0, 0.0, 0.0]
    rgba_camera_1stperson_spec.sensor_subtype = habitat_sim.SensorSubType.PINHOLE
    sensor_specs.append(rgba_camera_1stperson_spec)

    depth_camera_1stperson_spec = habitat_sim.SensorSpec()
    depth_camera_1stperson_spec.uuid = "depth_camera_1stperson"
    depth_camera_1stperson_spec.sensor_type = habitat_sim.SensorType.DEPTH
    depth_camera_1stperson_spec.resolution = camera_resolution
    depth_camera_1stperson_spec.postition = [0.0, 0.6, 0.0]
    depth_camera_1stperson_spec.orientation = [0.0, 0.0, 0.0]
    depth_camera_1stperson_spec.sensor_subtype = habitat_sim.SensorSubType.PINHOLE
    sensor_specs.append(depth_camera_1stperson_spec)

    rgba_camera_3rdperson_spec = habitat_sim.SensorSpec()
    rgba_camera_3rdperson_spec.uuid = "rgba_camera_3rdperson"
    rgba_camera_3rdperson_spec.sensor_type = habitat_sim.SensorType.COLOR
    rgba_camera_3rdperson_spec.resolution = camera_resolution
    rgba_camera_3rdperson_spec.postition = [0.0, 1.0, 0.3]
    rgba_camera_3rdperson_spec.orientation = [-45, 0.0, 0.0]
    rgba_camera_3rdperson_spec.sensor_subtype = habitat_sim.SensorSubType.PINHOLE
    sensor_specs.append(rgba_camera_3rdperson_spec)

    # agent configuration
    agent_cfg = habitat_sim.agent.AgentConfiguration()
    agent_cfg.sensor_specifications = sensor_specs

    return habitat_sim.Configuration(backend_cfg, [agent_cfg])

def simulate(sim, dt=1.0, get_frames=True):
    # simulate dt seconds at 60Hz to the nearest fixed timestep
    print("Simulating " + str(dt) + " world seconds.")
    observations = []
    start_time = sim.get_world_time()
    while sim.get_world_time() < start_time + dt:
        sim.step_physics(1.0 / 60.0)
        if get_frames:
            observations.append(sim.get_sensor_observations())

    return observations

if __name__ == "__main__":

    # create the simulators AND resets the simulator

    cfg = make_configuration()
    try:  # Got to make initialization idiot proof
        sim.close()
    except NameError:
        pass
    sim = habitat_sim.Simulator(cfg)
    agent_transform = place_agent(sim)

    # get the primitive assets attributes manager
    prim_templates_mgr = sim.get_asset_template_manager()

    # get the physics object attributes manager
    obj_templates_mgr = sim.get_object_template_manager()
    

    data_path = "xxxx/objects_v0.2"
    output_path = "xxxxx/output_add_object/"
    # load some object templates from configuration files
    sphere_template_id = obj_templates_mgr.load_configs(
        str(os.path.join(data_path, "banana"))
    )[0]
    print("sphere_template_id:",sphere_template_id)
    # add an object to the scene
    id_1 = sim.add_object(sphere_template_id)
    print("id_1:",id_1)
    orientation = mn.Quaternion.rotation( mn.Rad(math.pi / 2.0), np.array([-1.0, 0, 0]))

    sim.set_translation(np.array([2.50, 0, 0.2]), id_1)
    sim.set_rotation(orientation, id_1)
    print("id_1:",id_1)
    ipdb.set_trace()

    make_video = True
    show_video = False
    observations_list = []
    for i in range(30):
        # simulate
        observations = simulate(sim, dt=1, get_frames=make_video)
        sim.set_translation(sim.agents[0].get_state().position + np.array([0.01*i, 1.0, 0]), id_1)
        sim.set_rotation(mn.Quaternion.rotation( mn.Rad(math.pi / 2.0/6*i), np.array([-1.0, 0, 0])), id_1)
        observations_list.extend(observations)

    if make_video:
        sensor_dims = (
                sim.get_agent(0).agent_config.sensor_specifications[0].resolution
            )
        overlay_dims = (int(sensor_dims[1] / 4), int(sensor_dims[0] / 4))
        overlay_settings = [
            {
                "obs": "rgba_camera_1stperson",
                "type": "color",
                "dims": overlay_dims,
                "pos": (10, 10),
                "border": 2,
            },
            {
                "obs": "depth_camera_1stperson",
                "type": "depth",
                "dims": overlay_dims,
                "pos": (10, 30 + overlay_dims[1]),
                "border": 2,
            },
        ]

        vut.make_video(
            observations= observations_list,
            primary_obs="rgba_camera_3rdperson",
            primary_obs_type="color",
            video_file=output_path + "sim_basics",
            fps=60,
            open_vid=show_video,
            overlay_settings=overlay_settings,
            depth_clip=10.0,
        )

    print("OK")
```

aclegg3, 2022-01-04:
Hey @yyf17, sounds like 2 questions here.

1. How to add/change the material of an object.

aclegg3, 2022-01-04:
Hey @yyf17,

Sounds like to questions here:

1. How to modify the scale of the object? You can change the scale in the object's template/attributes metadata object before instantiation (see [the scale property](https://aihabitat.org/docs/habitat-sim/habitat_sim.attributes.ObjectAttributes.html#scale)).  Also see[ this tutorial's](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Advanced_Features.ipynb) `Object Template modification` sections for example code and more details.

2. How to add/change the material of an object? A programmatic material edit feature is not implemented currently in Habitat, so you won't be able to update this online in the simulator. However, you can modify the material referenced by the asset using Blender or some other external tool and then import the modified asset.

closed 2022-01-04 by aclegg3