facebookresearch/habitat-sim #1615: Math::Vector::ProjectedOntoNormalized nan

nikwalia, 2022-01-05:
I'm running into an issue when running Joel Ye's code for PointNav-Aux from without modifications (https://github.com/joel99/habitat-pointnav-aux), but it seems like the issues seem to be stemming from Habitat-Sim. I run into this error:

```
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 201, in <module>
    main()
  File "habitat_baselines/run.py", line 69, in main
    ckpt_wrapper(**vars(args))
  File "habitat_baselines/run.py", line 73, in ckpt_wrapper
    run_exp(exp_config, run_type, ckpt_path=ckpt_path, run_id=run_id, run_suffix=run_suffix, opts=opts)
  File "habitat_baselines/run.py", line 136, in run_exp
    trainer.train()
  File "/workspace/nikashw2/pointnav-2/habitat_baselines/rl/ppo/ppo_trainer.py", line 442, in train
    rollouts, current_episode_reward, running_episode_stats
  File "/workspace/nikashw2/pointnav-2/habitat_baselines/rl/ppo/ppo_trainer.py", line 254, in _collect_rollout_step
    outputs = self.envs.step([a[0].item() for a in actions])
  File "/workspace/nikashw2/pointnav-2/habitat/core/vector_env.py", line 345, in step
    return self.wait_step()
  File "/workspace/nikashw2/pointnav-2/habitat/core/vector_env.py", line 332, in wait_step
    observations.append(read_fn())
  File "/opt/conda/envs/habitat-pointnav/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/opt/conda/envs/habitat-pointnav/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/opt/conda/envs/habitat-pointnav/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError
```

I'm not sure what the source of the actual problem is. I'm only using the `RGB` and `POINTGOAL_WITH_GPS_COMPASS_SENSOR` sensors. I also worked through the dataset and eliminated episodes with bad data (as suggested in issue #692). Unfortunately, similarly to the previous issue, I am unable to provide exact reproducibility and the issue does not always occur.

I am currently using v0.1.7. For the repo in question, one would need the Gibson-Habitat train datasets; adjust paths and then run `./scripts/train_pn.sh baseline`.

nikwalia, 2022-01-09:
I thought that this issue might have been resolved by performing a clean dataset download. However, after a few runs, the same error has popped up again.

aclegg3, 2022-08-30:
Hey @nikwalia, this is likely an issue with specific older versions of habitat-sim and invalid agent positions in the dataset. You'll need to continue debugging to find the antecedents if this is a priority for you.

Closing because we can't really support issues with the episode itself or legacy versions of the simulator.

closed 2022-08-30 by aclegg3