facebookresearch/habitat-sim #1620: Incorrect rotation when setting agent to a pre-computed translation & rotation in Matterport3d

leobxpan, 2022-01-12:
#### Habitat-Sim version
v0.2.1

Following a [previous thread](https://github.com/facebookresearch/habitat-sim/issues/1608), I'm trying to set the agent to follow a pre-computed sequence of translations & rotations and save RGB observations (essentially I only care about the camera but not the agent, and the camera will be placed at these translations & rotations). However, while the translations look correct, the rotations seem to be off that the agent is looking towards the floor, while it's supposed to look at the wall. 

My code snippet:
```
    sim_settings = make_default_settings()
    sim_settings["sensor_pitch"] = 0
    sim_settings["sensor_height"] = 1. 
    
    make_simulator_from_settings(sim_settings)
    
    visual_sensor = sim._sensors["color_sensor_1st_person"]
    
    # set sensor initial position
    visual_sensor._spec.position = np.array([0, 0, 0]) 
    visual_sensor._spec.orientation = np.array([0, 0, 0]) 
    visual_sensor._sensor_object.set_transformation_from_spec()
    
    # load motion sequence
    motion_seq = np.load(args.npz_path)
    trans = motion_seq["joints"][:, 0, :]																
    root_ori = motion_seq["root_orient"].reshape(-1, 3, 3)		
   
    # correct rotation to -y as gravity axis
    quat_mp3d_to_habitat = quat_from_two_vectors(np.array([0, 0, -1]), geo.GRAVITY)
    rot_mat_mp3d_to_habitat= quaternion.as_rotation_matrix(quat_mp3d_to_habitat)
    
    for t in range(trans.shape[0]):
        camera_pos_t = trans[t]
        root_ori_t = root_ori[t]
        
        root_ori_t_x = -root_ori_t[:, 0]
        root_ori_t_y = root_ori_t[:, 2]
        root_ori_t_z = root_ori_t[:, 1]
        camera_rot_t = np.stack((root_ori_t_x, root_ori_t_y, root_ori_t_z), axis=1)

        camera_pos_habitat = rot_mat_mp3d_to_habitat @ camera_pos_t
        camera_rot_habitat = rot_mat_mp3d_to_habitat @ camera_rot_t
        
        sim.get_agent(0).scene_node.translation = camera_pos_habitat
        sim.get_agent(0).scene_node.rotation = mn.Quaternion.from_matrix(camera_rot_habitat)
        observation = habitat_sim.utils.viz_utils.observation_to_image(sim.get_sensor_observations()['color_sensor_1st_person'], \
        							observation_type='color')
```

This image shows the visualization of the sequence of `root_ori_t` and `camera_pos_t` in the original matterport scene. As can be seen the rotation points towards the wall.
[image]

The actual observations I'm getting from this snippet are like below, which seem to be right in terms of translation but wrong in rotation.
[image]

Any help would be much appreciated.

leobxpan, 2022-01-14:
Maybe @aclegg3 @erikwijmans? Any insights would be much appreciated.

erikwijmans, 2022-01-14:
What does `camera_rot_t` represent? Is it it transform points from camera space into world space or from world space into camera space? The rotation of the agent in habitat is the former -- camera space into world space -- so you may need to invert that matrix.

If you don't know, I'd try inverting and seeing what happens as everything else looks correct (assuming the axis swapping is correct).

leobxpan, 2022-01-14:
Thanks for your reply @erikwijmans `camera_rot_t` is actually world2cam. I inverted it but it's still off - now the rotation in x-y plane is inverted but the camera still points towards the floor.

The axis swapping should be correct. The first figure basically visualizes using `camera_rot_t` as the camera poses.

erikwijmans, 2022-01-15:
What directions do the axes of the camera represent? In habitat the camera looks along -Z, +Y is up, and +X is right.

leobxpan, 2022-01-15:
Yeah as you can see this `rot_mat_mp3d_to_habitat` accounts for this change and has been applied to both translation and rotation (works fine for translation). In the original mp3d frame the front vector is +Y, so theoretically after this correction it should be -Z.

leobxpan, 2022-01-15:
And sorry `camera_rot_t ` is actually cam2world (the camera pose), as can be seen from the first picture. So it shouldn't need an invertion.

erikwijmans, 2022-01-16:
I don't see anything wrong with the code so either I am missing some bug in the code or the axes in the rotation matrix don't represent what you think.

It's not uncommon for the rotation component to have a different coordinate convention from translation component (this is sometimes done to make interop with certain renders easier). I'd try just adding axis swap rotations to make things look correct for one image and see if that generalizes to other cases.

leobxpan, 2022-01-16:
Yeah I could just heuristically swap the axes to get reasonably looking observations. Was just worried if it's not following the correct logic and might cause problems later. I guess I'll go with this for now. Thanks for your help anyways.

closed 2022-01-16 by leobxpan