facebookresearch/habitat-sim #1622: Segmentation fault on M1 Mac

superMDguy, 2022-01-14:
## Habitat-Sim version
0.2.1 (tried installing in various ways)

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? The bug you are about to report may already be fixed in the latest version.

Main branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug

Segmentation fault occurs when trying to start a new `habitat_sim.Simulator`

## Steps to Reproduce

Steps to reproduce the behavior:

```
Python 3.9.9 | packaged by conda-forge | (main, Dec 20 2021, 02:41:07) 
[Clang 11.1.0 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import habitat_sim
>>> habitat_sim.Simulator()
[1]    15478 segmentation fault  python
```

(This is a minimal example, I tried running the [example notebook](https://github.com/facebookresearch/habitat-sim/blob/main/examples/tutorials/nb_python/ECCV_2020_Navigation.py) which is where it originally failed)

Installation doesn't work using `conda`, so I tried using both `pip` and `python setup.py install`, and neither worked

## Additional context

## System Info

```
ENVIRONMENT INFO:
Platform: macOS-12.0.1-arm64-arm-64bit
Machine: arm64
Processor: arm
Libc version:  
Mac version: 12.0.1
Python version: 3.9.9
Architecture: 64bit 
Win version:    
System OS: Darwin
Release: 21.1.0
Version: Darwin Kernel Version 21.1.0: Wed Oct 13 17:33:24 PDT 2021; root:xnu-8019.41.5~1/RELEASE_ARM64_T8101
Operational System: darwin
GCC version: b'Apple clang version 13.0.0 (clang-1300.0.29.30)\n'
CMAKE version: b'cmake version 3.22.1\n'
NVIDIA-SMI: b''
Pip packages versions:
b''
Conda packages versions:
b''```

erikwijmans, 2022-01-14:
Your repro segfault is due to a weird thing, python apparently still calls the destructor even if `__init__` errored. The simulator constructor requires a configuration.

That example works on my M1 tho. Do you have the scene downloaded? If not, you download with `python -m habitat_sim.utils.datasets_download --uids mp3d_example_scene`.

superMDguy, 2022-01-14:
I just downloaded the example scene and it's working now, thanks for the help!

closed 2022-01-14 by superMDguy