facebookresearch/habitat-sim #1627: Not able to load .glb file in habitat_sim.Simulator().

shaantanu314, 2022-01-19:
## Habitat-Sim version
v0.2.1
Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

I have some test scannet dataset which has been converted to .glb. when I'm constructing a simulator, the .glb file is not being loaded. The following is the complete error log 

Error Log:

> ---------------------------------------------------------------------------
> ValueError                                Traceback (most recent call last)
> <ipython-input-6-859693872908> in <module>
>      26     pass
>      27 try:
> ---> 28     sim = habitat_sim.Simulator(cfg)
>      29 except NameError:
>      30     print("error")
> 
> ~/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim/simulator.py in __init__(self, config, async_draw_agent_ids)
>       9     self._async_draw_agent_ids = async_draw_agent_ids
>      10     self._Simulator__last_state = __attr_factory__Simulator__last_state()
> ---> 11     self.__attrs_post_init__()
> 
> ~/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim/simulator.py in __attrs_post_init__(self)
>     119     def __attrs_post_init__(self) -> None:
>     120         self._sanitize_config(self.config)
> --> 121         self.__set_from_config(self.config)
>     122 
>     123     def close(self, destroy: bool = True) -> None:
> 
> ~/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim/simulator.py in __set_from_config(self, config)
>     250 
>     251     def __set_from_config(self, config: Configuration) -> None:
> --> 252         self._config_backend(config)
>     253         self._config_agents(config)
>     254         self._config_pathfinder(config)
> 
> ~/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim/simulator.py in _config_backend(self, config)
>     194     def _config_backend(self, config: Configuration) -> None:
>     195         if not self._initialized:
> --> 196             super().__init__(config.sim_cfg, config.metadata_mediator)
>     197             self._initialized = True
>     198         else:
> 
> ValueError: ::createSceneInstance : Cannot load: scene0010_01_vh_clean.glb
>

jturner65, 2022-01-19:
Hi!
Can you verify the integrity of the glb itself? (i.e. are you able to load it in blender, or some viewer such as https://gltf-viewer.donmccurdy.com/  ?)

shaantanu314, 2022-01-19:
yeah i think the assimp package was not installed properly. Thanks for responding

closed 2022-01-26 by Skylion007