facebookresearch/habitat-sim #1634: Determining available positions similar to THOR's GetReachablePositions

sagadre, 2022-01-23:
## ❓ Questions and Help
Is there any way to get all reachable positions in a navmesh (up to some discretization like `FORWARD_STEP_SIZE`).
Thanks!

erikwijmans, 2022-01-23:
Not really. There is `sim.pathfinder.get_topdown_view` which returns an occupancy grid which may be helpful for you use case but this isn't exactly all reachable positions. 

The set of all reachable positions can be extremely large -- e.g. an agent with a fairly small turn angle (say 30 degrees) can reach hundreds of millions unique locations even within just a 1 meter square even with a relatively big step size like 0.25m.

sagadre, 2022-02-03:
Am I right to think if I called `sim.pathfinder.get_topdown_view(meters_per_pixel=0.1,  height=0)` I would get an occupancy grid of floor "cells" at 0.1 meter voxel resolution?

erikwijmans, 2022-02-04:
The floor may not be at height zero. If you already have an agent position you can use its height, otherwise you can sample navigable locations via `sim.pathfinder.get_random_navigable_point`. Note that those may not always be on the "floor"

closed 2022-02-04 by sagadre