facebookresearch/habitat-sim #1641: Constrained look only works for a singular look action type

lucasosouza, 2022-01-28:
## Habitat-Sim version
v0.2.1

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

In `_rotate_local` there is a check to only allow constraints to be defined for a "singular look action type":

```
    if constraint is not None:
        rotation = scene_node.rotation

        if abs(float(rotation.angle())) > 0:
            ref_vector = mn.Vector3()
            ref_vector[axis] = 1

            if mn.math.angle(ref_vector, rotation.axis().normalized()) > mn.Rad(1e-3):
                raise RuntimeError(
                    "Constrained look only works for a singular look action type"
                )
```

What is the rationale behind this constraint? We are trying to implement a camera that can only rotate 90deg in either direction. There is a check in that same function that ensures whatever constraint defined in the actuation spec is observed, so the look_angle never goes beyond -90 or +90. But if we implement constraints for both look_left and look_right actions, look_left will work, but look_right will raise a runtime error once the camera's rotation angle is > 0 due to the exception defined above. 

We just started working with habitat_sim (awesome library), so apologies in advance if it has a simple explanation.

erikwijmans, 2022-01-29:
Looks like I missed an `abs` when writing that check. That check is there because the code doesn't handle constraining rotations with rotations along multiple axes because that gets tricky due to chained rotations.

lucasosouza, 2022-01-30:
Thanks Erik. That makes sense. Where would the missing `abs` go? It seems to me it should be applied to the rotation axis used in the assert. This works but I am wondering if there is a simpler version: `mn.Vector3(np.abs(rotation.axis().normalized()))`

erikwijmans, 2022-01-31:
I was thinking `abs(float(mn.math.angle(ref_vector, rotation.axis().normalized())))` but that works too

closed 2022-03-24 by aclegg3