facebookresearch/habitat-sim #1642: How could I make the poses and images been extracted  accurately matched ?

mhfy6868, 2022-01-29:
## Habitat-Sim version
v0.2.1

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/ Yes

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/ Yes

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
Hello, I've tried to extract images and correspondant poses  from Replica Data Set with image extractor, and would like to train them on nerf. 
[image]

I convert the poses with python script to transform them to the world coordinate system in order to meet the nerf requirements.
The poses from habitat-sim look like this:
[image]

However, I found the poses doesn't train well. Instead, the poses extracted from the images with colmap works perfectly.
And with colmap gui,  the colmap analyzed poses look like this:
[image]

I'm thinking if the image extractor api designed to only be able to extract images,semantic and depth information, but the matching of the poses feed to the extractor can not match the extract images,  which there might bring slight inaccuracy or approximations while the agent in the simulator is extracting images.  

As I mentioned that I post process the poses from Replica, but they are accurate and consistent as the poses been fed to the habitat-simulator. So, would there be inaccuracy or approximations in the image extracting process?  And how could I make the poses and images been extracted  accurately matched?   

Thanks in advance

erikwijmans, 2022-01-29:
The ImageExtractor can match poses exactly. It simply sets the agent position in a way that entirely ignores physics and thus is really just setting the extrinsic matrix used by the render.

One thing that is may be happening is that the ImageExtractor API is designed to work with agent positions, not camera positions. The camera is +1.5m along the z axis (in the agent's coordinate system): https://github.com/facebookresearch/habitat-sim/blob/4ac7c19e395c1d62b22d5bc58e47eaab3d36360c/src_python/habitat_sim/utils/data/data_extractor.py#L261

You can either just set that to zero in your local copy of the code or subtract `R @ [0, 0, 1.5]` from your positions where `R` is the corresponding rotation.

closed 2022-03-01 by aclegg3

derrick-xwp, 2022-06-28:
> ImageExtractor

One thing that is may be happening is that the ImageExtractor API is designed to work with agent positions, not camera positions. The camera is +1.5m along the z axis (in the agent's coordinate system): 

Questions:
1) What is the agent's coordinate system? is it opengl or opencv?
2) R @ [0, 0, 1.5], How can this  subtract 1.5?

oskar0812, 2023-04-12:
> > 图像提取器
> 
> 可能发生的一件事是 ImageExtractor API 设计用于处理代理位置，而不是摄像机位置。相机沿 z 轴 +1.5m（在代理的坐标系中）：
> 
> 问题：
> 
> 1. 代理的坐标系是什么？是 opengl 还是 opencv？
> 2. R@[0, 0, 1.5]，这个怎么减去1.5？

Hi, I was wondering if you solved this problem? I also encountered the same problem

oskar0812, 2023-04-12:
> One thing that is may be happening is that the ImageExtractor API is designed to work with agent positions, not camera positions. The camera is +1.5m along the z axis (in the agent's coordinate [...]

Hi, I was wondering if you solved this problem? I also encountered the same problem