facebookresearch/habitat-sim #1659: ArUco markers in Habitat Sim?

troiwill, 2022-02-09:
How can one add ArUco markers to Habitat Sim? I did not find any tutorials or discussions on how to add these markers.

aclegg3, 2022-02-09:
Hey @troiwill, there is no explicit support for fiducial markers in Habitat-sim. 

However, it should be trivial to add and managed these as [kinematic rigid objects](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedRigidObject.html). Checkout the `Advanced Topic : Object and Primitive Asset Customization` section of the [ECCV 2020: Habitat-sim Advanced Features](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Advanced_Features.ipynb) tutorial for concrete details on managing object assets.

 At a high level, I would:
 - Create a simple 2D square mesh and apply the fiducial texture in Blender or another 3D editor. 
 - Export each fiducial asset as a `.glb` file.
 - Load the objects into Habitat-sim as described in the tutorial linked above.
 - Instance the desired fiducial object into the world, set it as [MotionType::KINEMATIC](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.html#MotionType) and possibly [disable collisions](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedRigidObject.html#collidable).
 - Then you can move the object(s) anywhere in the scene where they will be visible by your camera sensors.

troiwill, 2022-02-12:
Hi @aclegg3, thanks for the help! I will check out this solution.