facebookresearch/habitat-sim #1672: Pathfinder not initialized for HM3D scene

arjung128, 2022-02-21:
## Habitat-Sim version
v0.2.1

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

When I run the [ECCV 2020 Navigation tutorial](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb) with the MP3D example scene, I'm able to generate the top-down map as expected. However, when I replace this for a scene from HM3D, specifically: `test_scene = "./data/scene_datasets/hm3d/train/00029-4wCTuaUNWEd/4wCTuaUNWEd.glb"`, I get `Pathfinder not initialized, aborting.` since `sim.pathfinder.is_loaded` is `False`.

Any ideas why this might be the case? Thanks!

arjung128, 2022-02-21:
For HM3D, you need to load the basis files, e.g `test_scene = "./data/scene_datasets/hm3d/train/00029-4wCTuaUNWEd/4wCTuaUNWEd.basis.glb"`

closed 2022-02-21 by arjung128

Eku127, 2024-10-29:
For habitat-sim 0.2.4 version, loading .glb w/o basis is ok, but for the 0.3.1 version the pathfinder will not initialized. For the 0.3.1 version using the .glb with basis one will work.