facebookresearch/habitat-sim #1673: URDF model with no collision

Zehao404, 2022-02-21:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
I load two urdf model in a scene but when I move them together, they will go through but not crashed. I enable physics and enable collision in urdf file. Is there some possible solution for solving this problem?

Zehao404, 2022-02-23:
Thank you, it helps a lot. I have another question about collision, how to make agent have collision with urdf model? Here is the urdf model I used: 
```xml
<?xml version="1.0" encoding="utf-8"?>

<robot
  name="buff1">
  <link name="base_link">
  </link>
  <link name="bottom">
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <box size="0.38 0.13 0.06"/>
      </geometry>
      <material name = "colour">
            <color rgba = "0 0 0 1"/>
        </material>
    </visual>
    <collision group="2">
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <box size="0.38 0.13 0.06"/>
      </geometry>
    </collision>
  </link>
  
  <joint name="base_link_bottom" type="fixed">
    <parent link="base_link"/>
    <child link="bottom"/>
    <origin xyz="0 0 0.03" rpy="0 0 0"/>
  </joint>
  
</robot>
```

Zehao404, 2022-02-24:
@aclegg3 We have met another problem, which is when we load a .glb model and set the motion type to dynamic, it can't fall to ground and stop at some distance from ground. When we set the height lower than this distance, it will be squeezed up and fall at that height again. We guess that is caused by navmesh, is it correct? Thank you!

aclegg3, 2022-02-24:
> `<collision group="2">`
This "group 2" flag is actually specifying a perf optimization for the simulator. Namely, that the link is a STATIC object and that you will load the asset with the fixed_base option or set it's MotionType to STATIC. This link won't collide with the background and will not automatically update its position from dynamics (it will update with explicit state setters).

If you want this URDF to be fully dynamic with a moving base then you should remove this tag, otherwise load it with fixed_base=true

aclegg3, 2022-02-24:
> @aclegg3 We have met another problem, which is when we load a .glb model and set the motion type to dynamic, it can't fall to ground and stop at some distance from ground. When we set the height [...]

This actually depends on the collision parameters, not the NavMesh. Bullet physics recommends a collision "margin" for stable collision detection/resolution. The default is 0.04 (4cm) for the STATIC environment mesh. There is also a margin on the objects, but I think the default is lower 0.01? for these. You can configure the margin in both the stage and object config files as well as the loaded [ObjectAttributes metadata](https://aihabitat.org/docs/habitat-sim/habitat_sim.attributes.ObjectAttributes.html#margin) before you instance the object. If you want centimeter accuracy on the dynamic resting state, playing with this parameter is one option.

Another option is to supply an additional collision asset, for example with lower floors than the render asset by a couple centimeters. This can be setup [in the config file for the stages](https://aihabitat.org/docs/habitat-sim/attributesJSON.html#stageattributes) and objects.

aclegg3, 2022-02-24:
BTW, using either the viewer.cpp or viewer.py applications for Habitat, you can use the `','` key to show a collision asset debug rendering. Issues with your collision assets may become more obvious upon inspection.

closed 2022-03-01 by aclegg3