facebookresearch/habitat-sim #1674: How to ignore the following warning?

yyf17, 2022-02-21:
```
W0221 16:34:14.326818 26834 ManagedPhysicsObjectBase.h:324] This object no longer exists. Please delete any variable references.
```
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.1

Habitat-Sim: v0.2.1

aclegg3, 2022-02-22:
Hey @yyf17,

This output indicates that you are holding and making API calls to a `ManagedObject` (e.g. a [ManagedRigidObject](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedRigidObject.html)) which has already been deleted or gone out of scope. Likely you are updating the position or state of a visualization object or something which is added and removed without the object reference being removed from a local cache in your script.

While I suggest you find this bug, if you just want to silence logger output, you can do so as described in our [configurable logging API docs](https://aihabitat.org/docs/habitat-sim/logging.html).

yyf17, 2022-02-25:
@aclegg3 Thank you！

closed 2022-02-25 by aclegg3