facebookresearch/habitat-sim #1675: The magnum package installed along with habitat-sim setup cannot find libOpenGL.so.0 in a local usr directory

ZJULiHongxin, 2022-02-21:
I tried to directly install habitat-sim using "conda install", but failed because libOpenGL.so.0 was not installed in the root/usr/lib. The error information is shown below.

ImportError: libOpenGL.so.0: cannot open shared object file: No such file or directory

SInce I had no right to install any software in root/usr, I created a local usr directory and put libOpenGL.so.0 in there. Its path was /home/username/my_usr/usr/lib/x86_64-linux-gnu/libOpenGL.so.0.

I also added these lines to ~/.bashrc and implemented "source ~/.bashrc"

- export PATH=$PATH:/home/username/my_usr/usr/
- export PATH=$PATH:/home/username/my_usr/usr/bin/
- export PATH=$PATH:/home/username/my_usr/usr/lib/x86_64-linux-gnu

after all of these, I installed habitat-lab and then habitat-sim, without magnum package being installed beforehand. However, the magnum package installed along with habitat-sim **failed to detect the libOpenGL.so.0 in my local usr directory.** 

**The same error occurred** when I tried to import habitat:
ImportError: libOpenGL.so.0: cannot open shared object file: No such file or directory

any solution?

Thanks!

aclegg3, 2022-02-28:
@erikwijmans @mosra thoughts?

ZJULiHongxin, 2022-03-01:
I asked the system admin to add me to the sudoer list, and then installed libopengl.so and some other trivial dependencies using sudo by myself.

Eventually, I reinstalled habitat-lab and then habitat-sim. It worked !

So this kind of proved my assumption that habitat-sim goes to check libOpenGL.so.0 only in the root/usr/lib when installing magnum.

erikwijmans, 2022-03-03:
No, habitat-sim just uses the dynamic loader so by extension it'll check wherever that checks. The correct environment variable to add things to the dynamic loader's search path is `LD_LIBRARY_PATH` not `PATH.

ZJULiHongxin, 2022-03-04:
Thanks a lot for your solution @erikwijmans! I changed `PATH` to `LD_LIBRARY_PATH`, and got everything done.

closed 2022-03-04 by ZJULiHongxin

TopCoder2K, 2022-03-30:
@ZJULiHongxin, could you please explain what needs to be done in more detail?
I have the same error and I can't ask sudo. Where do I need to add `LD_LIBRARY_PATH`? 
If I run `export PATH=$LD_LIBRARY_PATH`, I even can't run python because it's not found

erikwijmans, 2022-03-30:
`LD_LIBRARY_PATH` is the environment variable to set, not the value to set `PATH` to. So you need to do `export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/path/to/lib` to add the path to a library.

TopCoder2K, 2022-03-31:
@erikwijmans, @ZJULiHongxin thank you for the answers!

I created the same directories inside `/home/my_username`, downloaded the library from https://ubuntu.pkgs.org/20.04/ubuntu-main-amd64/libopengl0_1.3.1-1_amd64.deb.html and placed it in the deepest folder. So, the library's path is `/home/my_username/my_user/usr/bin/x86_64-linux-gnu/libopengl0_1.3.1-1_amd64.deb`. 
Then I added the lines into `~/.bashrc` changing 'username' to 'my_username' and ran `source ~/.bashrc`. The error remained. (I didn't reinstall habitat-sim, do I need to?)
What did I do wrong?

ZJULiHongxin, 2022-03-31:
@TopCoder2K Hey, bro! It seems that you have not got a clue about how to install the software in a local "usr"directory.

Firstly, you download "*.deb" in any folder.
`apt-get download your-package-name`

Secondly, you install it locally using dpkg.
`dpkg -x *.deb /home/username/my_usr`
Then you will see the installation results in /home/username/my_usr/usr/bin/xxx

At last, append this line to "~/.bashrc".
`export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/home/username/my_usr/usr/bin/`

Then, implement "source ~/.bashrc" to make it take effect.

Hope this is useful to you.

TopCoder2K, 2022-03-31:
OMG, really, ahahahaha. I've been using conda and pip too much that I even forgot how to work with *.deb packages...

I followed the steps above with the only note that I appended the previous 3 lines, not `export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/home/username/my_usr/usr/bin/` and it worked! 

Thank you, @ZJULiHongxin!