facebookresearch/habitat-sim #1677: Camera Intrinsic Parameters in View, Transform and Warp tutorial

arjung128, 2022-02-28:
## Habitat-Sim version
v0.2.1

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

From the [View, Transform and Warp tutorial](https://aihabitat.org/docs/habitat-lab/view-transform-warp.html):
```
K = np.array([
    [1 / np.tan(hfov / 2.), 0., 0., 0.],
    [0., 1 / np.tan(hfov / 2.), 0., 0.],
    [0., 0.,  1, 0],
    [0., 0., 0, 1]])
```
We know that the first entry is `f`, and also that `f = (W/2)/tan(hfov/2)`. Why is `W/2 = 1` here? Is this because `W=2` in the camera coordinate frame? But then does this mean we can always set `W/2 = 1` by picking our camera coordinate frame appropriately?

aclegg3, 2022-02-28:
Correct, the chosen camera coordinate space is [-1,1], so W==2.

@erikwijmans anything to add?

arjung128, 2022-03-01:
Got it, thanks!

closed 2022-03-01 by arjung128