facebookresearch/habitat-sim #1687: Collision detection

Franc1sNing, 2022-03-08:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
I implemented a collision of the agent through navmesh.
I try to use ContactPointData to detect the collision, but failed.
Are there have any other method to detect the collision which caused by navmesh?
Or check if navmesh is working

aclegg3, 2022-03-08:
Hey @Franc1sNing, 

The NavMesh is not really related to 3D collision detection via Bullet. You should consider these to be separate systems.

For more details and interactive examples, see our[ ECCV 2020 tutorials](https://aihabitat.org/tutorial/2020/).

 - To learn more about the NavMesh and how to use it to constrain agent motion, see the [Habitat-Sim Basics for Navigation ](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb) tutorial section.

 - To learn more about our Bullet physics integration and 3D collision queries, see the [Habitat-Sim for Interaction](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Interactivity.ipynb) tutorial section.

You may want to play around with these features interactively to get a sense for what is or isn't working for you. We have interactive `viewer` applications in both C++ and python for this purpose. See the [Testing section of the README](https://github.com/facebookresearch/habitat-sim#testing) for details.

closed 2022-03-08 by aclegg3

aclegg3, 2022-03-14:
> First How can i detect the obstruction effect created by navmesh?

All the NavMesh relevant functions are all explained in the ECCV tutorial. This includes querying whether a particular point is navigable or not. The final cell shows how we detect collisions between the agent and NavMesh. TLDR; Try to move the agent, run the NavMesh step filter, check whether the full movement was realized or cut short.

> Second I have observed that when the agent(loaded via urdf) appears next to an object (loaded via urdf) without touching the object, the object is displaced by millimeters. How can i fixed it?

There is a collision margin for each object which is on the order of centimeters by default and collision shapes may be approximations (e.g. convex hulls) of objects. I wouldn't expect millimeter accuracy unless you carefully tune your assets and parameters. You can use the viewer applications to visualize the collision shapes with `','`.

Franc1sNing, 2022-03-15:
Sorry for my carelessness, the first problem was solved .

Regarding the second question, I am sorry that it may be my description. The current situation is that the collision margin is in centimeters, but my agent is decimeters away from the object, and the object is continuously displaced in millimeters. Note It is continuous and does not stop, which means that the object will have a great displacement under the action of no external force for a long time. And not just the objects closest to me, all objects loaded through urdf in the scene are constantly undergoing such displacement.

抄送: "Ning ***@***.******@***.***&gt;;

aclegg3, 2022-03-15:
Interesting, I guess it may be hard to understand without seeing more details. 

You are welcome to upload images/videos and other media here. Alternatively, for higher frequency communication you are welcome to reach out in `general` or DM me on the Habitat community Slack workspace as described in the website [Contact section](https://aihabitat.org/#contact).

Franc1sNing, 2022-03-16:
Sorry my Slack has some problem, So I still have to contact you via email
I will send you a video. In the video, the left one is printing the pose and orientation of cube_1, left two is cube_2
Thank you for being patient with the video so you can spot the problem

抄送: "Ning ***@***.******@***.***&gt;;

 

 
 

从QQ邮箱发来的超大附件

1.mp4 (204.53M, 2022年04月15日 17:04 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=5b653933484652c0447878731132571c51500b0153045d0a4c540c05041f5103555c140a0101001e51035b05010b5d0a02040f55373765024f084907370f&code=ae9372e3

dhruvbatra, 2022-03-17:
Unfortunately, this is beyond the level of support we can provide.

Franc1sNing, 2022-03-18:
ok i got it. 
I have another question about ContactPointData, I want to know if it can detect when my agent is in contact with the scene. 
Just like Ray, shoot lasers and interact with the scene.
thanks for all your help

抄送: "Ning ***@***.******@***.***&gt;;

aclegg3, 2022-03-23:
> Just like Ray, shoot lasers and interact with the scene.

There is [raycasting](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#cast_ray) available within the simulator for testing against collision objects.