facebookresearch/habitat-sim #1689: ShortestPathFollower vs ShortestPath

tekinengin, 2022-03-11:
i have an issue with GreedyGeodesicFollower similar to #1497 . I find the shortest path from agent to a location using habitat_sim.ShortestPath. But when i try to get action by using ShortestPathFollower it gives _stop_on_error error. 

Agent action space contains forward, turn_right and turn_left. I tried with very fine actions (FORWARD_STEP_SIZE:0.05 and TURN_ANGLE:1) but it still behaves same. 

In below example i spawn the agent to a certain location to reproduce error. 

Habitat-sim : 0.1.6, I know it is littler old but i did not see any fix on this matter in later versions.
Gibson Scene: Sherrill

What would be the way to solve this issue?

[image]

erikwijmans, 2022-03-11:
Does it fail consistently or just once in a while? The follower is just a bunch of heuristics, not a full planner in action space (the latter is way too slow). It failing is just an unfortunate reality.

tekinengin, 2022-03-11:
It happens once in a while. Then i will settle with terminating sim and ignoring episode if i encounter this.

closed 2022-03-11 by tekinengin