facebookresearch/habitat-sim #1691: ImportError: cannot import name 'FetchRobotNoWheels' from 'habitat_sim.robots.fetch_robot' When Installed Via Conda

Hamptonjc, 2022-03-13:
## Habitat-Sim version
vx.x.x or main?
v0.2.1

## 🐛 Bug
cannot import FetchRobotNoWheels from habitat_sim.robots.fetch_robot after installation from conda. This is causing an issue for me when trying to use Habitat-Lab.

## Steps to Reproduce

Steps to reproduce the behavior:

1. conda create -n habitat python=3.7
2. conda activate habitat
3. conda install habitat-sim withbullet -c conda-forge -c aihabitat
4. python -c "from habitat_sim.robots.fetch_robot import FetchRobotNoWheels"

## Expected behavior

Import without error.

## Additional context

## System Info
`
ENVIRONMENT INFO:
Platform: Linux-5.4.0-104-generic-x86_64-with-debian-buster-sid
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.17
Mac version:
Python version: 3.7.11
Architecture: 64bit
Win version:
System OS: Linux
Release: 5.4.0-104-generic
Version: #118~18.04.1-Ubuntu SMP Thu Mar 3 13:53:15 UTC 2022
Operational System: linux
GCC version: b'gcc (Ubuntu 8.4.0-1ubuntu1~18.04) 8.4.0\n'
CMAKE version: b'cmake version 3.17.2\n'
NVIDIA-SMI: b'Sat Mar 12 22:30:51 2022       \n+-----------------------------------------------------------------------------+\n| NVIDIA-SMI 455.45.01    Driver Version: 455.45.01    CUDA Version: 11.1     |\n|-------------------------------+----------------------+----------------------+\n| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |\n| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |\n|                               |                      |               MIG M. |\n|===============================+======================+======================|\n|   0  GeForce GTX 106...  On   | 00000000:27:00.0 Off |                  N/A |\n| 33%   27C    P8    10W / 120W |    125MiB /  6077MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n'
Pip packages versions:
b''
Conda packages versions:
b''
`

erikwijmans, 2022-03-13:
Please install the nightly build with `-c aihabitat-nightly` as we haven't released a version with that functionality yet.

closed 2022-03-13 by erikwijmans

Hamptonjc, 2022-03-13:
@erikwijmans Nightly build fails to install:

``` 
$ conda install habitat-sim withbullet -c conda-forge -c aihabitat-nightly
...
UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package habitat-sim-mutex conflicts for:
habitat-sim -> habitat-sim-mutex==1.0[build='display_nobullet|headless_nobullet']
withbullet -> habitat-sim-mutex==1.0[build=*_bullet]The following specifications were found to be incompatible with your system:

  - feature:/linux-64::__glibc==2.27=0
  - feature:|@/linux-64::__glibc==2.27=0

Your installed version is: 2.27
```

erikwijmans, 2022-03-14:
Looks like our nightly builder is not working for some reason. You can try building from source in the mean time.

MYM181, 2023-12-24:
I encounter the same question now. Could you please tell me how you figure it out? Thanks!!

MYM181, 2023-12-24:
> ## Habitat-Sim version
> vx.x.x or main? v0.2.1
> 
> ## 🐛 Bug
> cannot import FetchRobotNoWheels from habitat_sim.robots.fetch_robot after installation from conda. This is causing an issue for me when trying to use Habitat-Lab. (i.e. this error is thrown when I import habitat lab v0.2.1 with `python -c "import habitat"`)
> 
> ## Steps to Reproduce
> Steps to reproduce the behavior:
> 
> 1. conda create -n habitat python=3.7
> 2. conda activate habitat
> 3. conda install habitat-sim withbullet -c conda-forge -c aihabitat
> 4. python -c "from habitat_sim.robots.fetch_robot import FetchRobotNoWheels"
> 
> ## Expected behavior
> Import without error.
> 
> ## Additional context
> ## System Info
> `ENVIRONMENT INFO: Platform: Linux-5.4.0-104-generic-x86_64-with-debian-buster-sid Machine: x86_64 Processor: x86_64 Libc version: glibc 2.17 Mac version: Python version: 3.7.11 Architecture: 64bit Win version: System OS: Linux Release: 5.4.0-104-generic Version: #118~18.04.1-Ubuntu SMP Thu Mar 3 13:53:15 UTC 2022 Operational System: linux GCC version: b'gcc (Ubuntu 8.4.0-1ubuntu1~18.04) 8.4.0\n' CMAKE version: b'cmake version 3.17.2\n' NVIDIA-SMI: b'Sat Mar 12 22:30:51 2022 \n+-----------------------------------------------------------------------------+\n| NVIDIA-SMI 455.45.01 Driver Version: 455.45.01 CUDA Version: 11.1 |\n|-------------------------------+----------------------+----------------------+\n| GPU Name Persistence-M| Bus-Id Disp.A | Volatile Uncorr. ECC |\n| Fan Temp Perf Pwr:Usage/Cap| Memory-Usage | GPU-Util Compute M. |\n| | | MIG M. |\n|===============================+======================+======================|\n| 0 GeForce GTX 106... On | 00000000:27:00.0 Off | N/A |\n| 33% 27C P8 10W / 120W | 125MiB / 6077MiB | 0% Default |\n| | | N/A |\n+-------------------------------+----------------------+----------------------+\n' Pip packages versions: b'' Conda packages versions: b''`

I encounter the same question now. Could you please tell me how you figure it out? Thanks!!