facebookresearch/habitat-sim #1693: Cannot run 'Habitat-sim Basics for Navigation' Colab (Installation cell not working)

yewonbyun, 2022-03-13:
## Habitat-Sim version
vx.x.x or main?

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? The bug you are about to report may already be fixed in the latest version.

Main branch contains 'bleeding edge' code, but we do appreciate bug reports for it! 

## 🐛 Bug

Cannot run 'Habitat-sim Basics for Navigation' Colab. Fails at first cell (!curl -L https://raw.githubusercontent.com/facebookresearch/habitat-sim/main/examples/colab_utils/colab_install.sh | NIGHTLY=true bash -s). Multiple conflict errors found. 

Error Message:

UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (python): ...
An error occured during the installation of Habitat-sim or Habitat-Lab.

## Steps to Reproduce

Steps to reproduce the behavior:

1. Open 'Habitat-sim Basics for Navigation' Colab
2. !curl -L https://raw.githubusercontent.com/facebookresearch/habitat-sim/main/examples/colab_utils/colab_install.sh | NIGHTLY=true bash -s

 Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (python):

  - cffi=1.15.0 -> pycparser -> python[version='2.7.*|2.7.*|>=3.4|>=3.6,<3.7.0a0|>=2.7,<2.8.0a0|3.6.*|3.5.*|3.4.*|>=3.6|>=3.5,<3.6.0a0|3.8.*|3.7.*|3.9.*|3.10.*']
  - cffi=1.15.0 -> python[version='>=3.10,<3.11.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0']
  - habitat-sim -> attrs[version='>=19.1.0'] -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*|>=3.6,<3.7.0a0|>=2.7,<2.8.0a0|>=3.7|>=3.4|>=3|3.4.*|>=3.5|>=3.5,<3.6.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0|>=3.10,<3.11.0a0|3.7.*|>=2.7']
  - habitat-sim -> python[version='>=3.7,<3.8.0a0']
  - numba=0.51.2 -> llvmlite[version='>=0.34.0,<0.35'] -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.10,<3.11.0a0|>=3.9,<3.10.0a0|>=3.5,<3.6.0a0|3.4.*|3.8.*|3.7.*']
  - numba=0.51.2 -> python[version='>=3.6,<3.7.0a0|>=3.8,<3.9.0a0|>=3.7,<3.8.0a0']
  - numpy=1.21.5 -> python[version='>=3.10,<3.11.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0']
  - numpy=1.21.5 -> python_abi=3.10[build=*_cp310] -> python[version='3.10.*|3.8.*|3.9.*|3.7.*']
  - pillow=7.1.2 -> olefile -> python[version='2.7.*|3.5.*|3.6.*|3.4.*|>=3.5,<3.6.0a0|>=2.7,<2.8.0a0|3.8.*|3.7.*']
  - pillow=7.1.2 -> python[version='>=3.6,<3.7.0a0|>=3.8,<3.9.0a0|>=3.7,<3.8.0a0']
  - python=3.7
  - scipy=1.4.1 -> numpy[version='>=1.18.1,<2.0a0'] -> python[version='3.6.*|>=2.7,<2.8.0a0|>=3.10,<3.11.0a0|>=3.9,<3.10.0a0|>=3.5,<3.6.0a0|3.7.*|3.8.*']
  - scipy=1.4.1 -> python[version='>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0']

The following specifications were found to be incompatible with each other:...

## Expected behavior

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/main/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python src_python/habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux):
 - How you installed PyTorch (`conda`, `pip`, source):
 - Build command you used (if compiling from source):
 - Python version:
 - GPU models and configuration:
 - CUDA version:
 - PIP/conda dependencies packages versions
 - Any other relevant information: Run on Colab (Runtime - GPU)

aclegg3, 2022-03-14:
Hey, @yewonbyun,

Sorry this didn't work for you yesterday. Sometimes the Colabs can be finicky when intermittent outages cause some resources to be unreachable or dependency versions conflict. Can you tray again now after a factory reset of the environment? I was able to successfully run this on Chrome today.

Note: cell 3 may crash with `module 'PIL.TiffTags' has no attribute 'IFD'`
In this case reset the runtime from the Colab menu `runtime->Reset runtime` and then run all again to clear the error.

closed 2022-03-24 by aclegg3

CHOcho-quan, 2022-03-25:
> Hey, @yewonbyun,
>
> Sorry this didn't work for you yesterday. Sometimes the Colabs can be finicky when intermittent outages cause some resources to be unreachable or dependency versions conflict. Can [...]

I cant resolve the case even reset the runtime. Are there any reason for the crash bug?

jayeshkumar1734, 2023-01-19:
Even after running the curl command, import habitat-sim gives error. Dont know whats the issue here