facebookresearch/habitat-sim #1696: Conda install with cuda

Hoyyyaard, 2022-03-18:
How can I install haibitat-sim through conda with cuda?
It doesnt work when I install with headless and It will report"must build habitat-sim with cuda for gpu2gpu-transfer"

dhruvbatra, 2022-03-18:
Too little information to help you. Please consider asking on slack to the community.

closed 2022-03-18 by dhruvbatra

Zhongxyyyy, 2022-05-11:
> How can I install haibitat-sim through conda with cuda? It doesnt work when I install with headless and It will report"must build habitat-sim with cuda for gpu2gpu-transfer"

Have you found the solution? I met the same question as well

Zhongxyyyy, 2022-05-13:
> "must build habitat-sim with cuda for gpu2gpu-transfer"

Thanks for you reply. I installed the habitat-sim with conda headless, but when I run task relaed code, it will report"must build habitat-sim with cuda for gpu2gpu-transfer". So should I need install from source and add ''--with-cuda'' to solve it?

erikwijmans, 2022-05-13:
Yep

Zhongxyyyy, 2022-05-26:
> Yep

Unfortunately, it didn't work in my computer, I will open a new issue to ask for help.

xwaiyy123, 2022-07-01:
> > Yep
> 
> Unfortunately, it didn't work in my computer, I will open a new issue to ask for help.

我遇到了相同的问题，这个问题你解决了吗？

Zhongxyyyy, 2022-07-06:
> 
解决了，最终是用source来build的，本地机器一直没弄好，用远程的服务器重建了一个环境搞好了

Zhongxyyyy, 2022-07-06:
> > > Yep
> > 
> > 
> > Unfortunately, it didn't work in my computer, I will open a new issue to ask for help.
> 
> 我遇到了相同的问题，这个问题你解决了吗？

解决了，最终是用source来build的，本地机器一直没弄好，用远程的服务器重建了一个环境搞好了

labixiaochao, 2025-07-28:
> > > > Yep
> > > 
> > > 
> > > Unfortunately, it didn't work in my computer, I will open a new issue to ask for help.
> > 
> > 
> > 我遇到了相同的问题，这个问题你解决了吗？
> 
> 解决了，最终是用source来build的，本地机器一直没弄好，用远程的服务器重建了一个环境搞好了

I used this command conda install habitat-sim=0.2.4 withbullet -c conda-forge -c aihabitat to install habitat-sim, but encountered the same error as yours when setting the gpu_gpu parameter. May I ask how you solved it? File "/home/simu/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.2.4-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 609, in _initialize_sensor
assert cuda_enabled, "Must build habitat sim with cuda for gpu2gpu-transfer"
AssertionError: Must build habitat sim with cuda for gpu2gpu-transfer
Error executing job with overrides: []
Traceback (most recent call last):
File "/home/simu/vln/vlfm/vlfm/run.p