facebookresearch/habitat-sim #1701: Error downloading HM3D dataset

ethanabrooks, 2022-03-21:
## Habitat-Sim version
v0.2.1

## 🐛 Bug

## Steps to Reproduce
```
python -m habitat_sim.utils.datasets_download --username $API_TOKEN_ID --password $API_TOKEN_SECRET --uids hm3d_minival
```
This outputs:
```
No data-path provided, default to: /home/ethanbro/generalization/data. Use '--data-path' to specify another location.
** Resuming transfer from byte position 10618
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    82  100    82    0     0   1108      0 --:--:-- --:--:-- --:--:--  1108
  0 10618    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
Traceback (most recent call last):
  File "/home/ethanbro/miniconda3/envs/habitat/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/ethanbro/miniconda3/envs/habitat/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/ethanbro/miniconda3/envs/habitat/lib/python3.8/site-packages/habitat_sim/utils/datasets_download.py", line 578, in <module>
    main(sys.argv[1:])
  File "/home/ethanbro/miniconda3/envs/habitat/lib/python3.8/site-packages/habitat_sim/utils/datasets_download.py", line 572, in main
    download_and_place(
  File "/home/ethanbro/miniconda3/envs/habitat/lib/python3.8/site-packages/habitat_sim/utils/datasets_download.py", line 410, in download_and_place
    with tarfile.open(data_path + package_name, "r:*") as tar_ref:
  File "/home/ethanbro/miniconda3/envs/habitat/lib/python3.8/tarfile.py", line 1608, in open
    raise ReadError("file could not be opened successfully")
tarfile.ReadError: file could not be opened successfully
```
## Expected behavior
Downloads minival split of HM3D data

## Additional context

## System Info

 - OS (e.g., Linux): Linux
 - Python version: python3.8
 - Any other relevant information:
Content 
<details>
  <summary>Content of `hm3d-minival-habitat.tar`</summary>

```
<!doctype html><html class="no-js" lang="en">
        <head>
        <script>window.CLOUD_UI_BASE_URL = "https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/";</script><meta charset="utf-8"/><title>MyMatterport</title><meta name="description" content=""/><meta name="viewport" content="width=device-width,initial-scale=1"/><meta name="referrer" content="origin"/><base href="/"/><meta name="importmap-type" content="systemjs-importmap"/><script type="systemjs-importmap" src="https://static.matterport.com/ui/import-map/22.3.1-22-g2b2608dd/import-map.json"></script><script>(function () {
        // Create a queue, but don't obliterate an existing one!
        var analytics = (window.analytics = window.analytics || []);

        // If the real analytics.js is already on the page return.
        if (analytics.initialize) return;

        // If the snippet was invoked already show an error.
        if (analytics.invoked) {
          if (window.console && console.error) {
            console.error('Segment snippet included twice.');
          }
          return;
        }

        // Invoked flag, to make sure the snippet
        // is never invoked twice.
        analytics.invoked = true;

        // A list of the methods in Analytics.js to stub.
        analytics.methods = [
[... 47 lines cut ...]
          // Insert our script next to the first script element.
          var first = document.getElementsByTagName('script')[0];
          first.parentNode.insertBefore(script, first);
        };

        // Add a version to keep track of what's in the wild.
        analytics.SNIPPET_VERSION = '4.0.0';
      })();</script><script src="https://cdn.jsdelivr.net/npm/promise-polyfill@8/dist/polyfill.min.js"></script><script src="https://cdn.jsdelivr.net/npm/systemjs/dist/system.js"></script><script src="https://cdn.jsdelivr.net/npm/systemjs@6.4.0/dist/extras/named-exports.min.js"></script><link rel="apple-touch-icon" sizes="57x57" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-57x57.png"/><link rel="apple-touch-icon" sizes="60x60" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-60x60.png"/><link rel="apple-touch-icon" sizes="72x72" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-72x72.png"/><link rel="apple-touch-icon" sizes="76x76" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-76x76.png"/><link rel="apple-touch-icon" sizes="114x114" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-114x114.png"/><link rel="apple-touch-icon" sizes="120x120" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-120x120.png"/><link rel="apple-touch-icon" sizes="144x144" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-144x144.png"/><link rel="apple-touch-icon" sizes="152x152" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-152x152.png"/><link rel="apple-touch-icon" sizes="180x180" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-180x180.png"/><link rel="apple-touch-icon" sizes="512x512" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/apple-touch-icon-512x512.png"/><link rel="icon" type="image/png" sizes="16x16" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/favicon-16x16.png"/><link rel="icon" type="image/png" sizes="32x32" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/favicon-32x32.png"/><link rel="icon" type="image/png" sizes="96x96" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/favicon-96x96.png"/><link rel="manifest" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/manifest.json"/><link rel="mask-icon" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/images/favicons/safari-pinned-tab.svg" color="#ff3158"/><meta name="msapplication-TileColor" content="#ff3158"/><meta name="theme-color" content="#ffffff"/><script>// Safari, in Private Browsing Mode, looks like it supports localStorage but all calls to setItem
      // throw QuotaExceededError. We're going to detect this and just silently drop any calls to setItem
      // to avoid the entire page breaking, without having to do a check at each usage of Storage.
      if (typeof localStorage === 'object') {
        try {
          localStorage.setItem('localStorage', 1);
          localStorage.removeItem('localStorage');
        } catch (e) {
          Storage.prototype._setItem = Storage.prototype.setItem;
          Storage.prototype.setItem = function () {};
          alert(
            'Your web browser does not support storing settings locally. In Safari, the most common cause of this is using "Private Browsing Mode". Some settings may not save or some features may not work properly for you.'
          );
        }
      }</script><link rel="stylesheet" href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/styles/base.css"/><script defer="defer" src="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/legacy_styles.js"></script><script defer="defer" src="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/legacy_vendor.js"></script><script defer="defer" src="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/legacy_app.js"></script><script defer="defer" src="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/legacy_templates.js"></script><script defer="defer" src="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/main.js"></script><link href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/legacy_styles.css" rel="stylesheet"><link href="https://static.matterport.com/cloud_ui/22.3.1-38-g013e3af25/main.css" rel="stylesheet"></head><body><div ng-attr-id="{{ viewName || 'matterport' }}" class="body" ng-app="mpwebApp" ng-controller="MasterCtrl" ng-class="{menuopen:headerMenuToggle}" ng-cloak><div class="wrapper" ng-class="{'capturesvc-intro-bg': viewName === 'capture_services' || viewName === 'capture_services_unavailable' || viewName === 'capture_services_available'}"><div id="mainContainer"><div id="skip"><a href="https://static.matterpor [... cut ...] ettings|billingsettings|shopsubscriptions|usagesettings|rolessettings|devtoolsettings" ng-switch-when-separator="|" mp-layout="collapsed" right-aligned></mp-footer><mp-footer ng-switch-default mp-layout="collapsed"></mp-footer></ng-switch></ng-container></div><script defer="defer" src="https://cdn.jsdelivr.net/npm/systemjs@6.8.3/dist/extras/amd.min.js"></script><script defer="defer" src="https://unpkg.com/systemjs/dist/extras/named-register.js"></script></div></div></body></html>
```

</details>

erikwijmans, 2022-03-21:
I was able to download all these. I'd just try again.

closed 2022-03-21 by erikwijmans

ethanabrooks, 2022-03-21:
I will continue to try to debug this on my own, but for the record, the command still produces the same error for me.

dhruvbatra, 2022-03-22:
Are you sure you received access to HM3D scenes from Matterport? Basically, steps at the bottom of page here: https://matterport.com/partners/facebook

JanTanja, 2022-03-22:
Getting the same thing as @ethanabrooks

ethanabrooks, 2022-03-22:
After contacting Matterport customer support, I received an email from them indicating that the research dataset was only available under "paid subscription plans,, Starter or above." I haven't gotten such a plan and verified that the download worked afterword but I suspect this is the cause of the issue.

It would be nice if this requirement was advertised somewhere, either on the Matterport "developers tools" page, the error that I printed, or the Habitat docs.

dhruvbatra, 2022-03-22:
Thanks for letting us know. 

I just spoke to my Matterport colleague. They assured me that the dataset is available for free and not contingent on a paid plan. They believe it was a miscommunication by the customer support team.

ethanabrooks, 2022-03-22:
What do you suggest as the next step for us? Here is a screenshot of the email if it's helpful:
[image]

dhruvbatra, 2022-03-23:
Sorry you're stuck in this situation. I recommend email them back and pointing to their own website: https://matterport.com/partners/facebook

wz0919, 2022-04-28:
Same issue here, have you debugged it?

ethanabrooks, 2022-04-28:
Unfortunately, I ended up moving on to a different project before sorting this out. Good luck though.

erikwijmans, 2022-04-30:
There shouldn't be any issue here. One possibly confusing thing is that the dataset access and API token creation is in the developer tools settings. The top of that page talks about paid accounts, but you don't need one. You can scroll down to the bottom to request access to the HM3D dataset and you can generate an API token on a free account.

Rh-Dang, 2024-05-23:
I solved the problem. I first applied for access to HM3D and then re-generated the token on the website. With the new token, the above problems do not occur.

wangyu-ustc, 2025-02-08:
> I solved the problem. I first applied for access to HM3D and then re-generated the token on the website. With the new token, the above problems do not occur.

I did the same thing but am still encountering this issue