facebookresearch/habitat-sim #1703: Got some errors when install the CUDA version from source.

Mingxiao-Li, 2022-03-23:
Hi, When I run "python setup.py install --headless --with-cuda --cmake  " I always got the below error : 

running install                                                                                                            
running bdist_egg                                                                                                          
running egg_info                                                                                                           
writing habitat_sim.egg-info/PKG-INFO                                                                                      
writing dependency_links to habitat_sim.egg-info/dependency_links.txt                                                      
writing requirements to habitat_sim.egg-info/requires.txt                                                                  
writing top-level names to habitat_sim.egg-info/top_level.txt                                                              
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'                                                                   
reading manifest template 'MANIFEST.in'                                                                                    
adding license file 'LICENSE'                                                                                              
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'                                                                   
installing library code to build/bdist.linux-x86_64/egg                                                                    
running install_lib                                                                                                        
running build_py                                                                                                           
running build_ext                                                                                                          
-- The CUDA compiler identification is unknown                                                                             
-- Detecting CUDA compiler ABI info                                                                                        
-- Detecting CUDA compiler ABI info - failed                                                                               
-- Check for working CUDA compiler: /usr/local/cuda                                                                        
-- Check for working CUDA compiler: /usr/local/cuda - broken                                                               
CMake Error at /data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/site-packages/cmake/data/share/cmake-3.18/Module
s/CMakeTestCUDACompiler.cmake:52 (message):                                                                                
  The CUDA compiler                                                                                                        
                                                                                                                           
    "/usr/local/cuda"                                                                                                      
                                                                                                                           
  is not able to compile a simple test program.                                                                            
                                                                                                                           
  It fails with the following output:                                                                                      
                                                                                                                           
    Change Dir: /data/leuven/333/vsc33366/projects/habitat-sim/build/CMakeFiles/CMakeTmp                                   
                                                                                                                           
    Run Build Command(s):/usr/bin/gmake cmTC_e7ae4/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_e7ae4.dir/build.make CMakeFil
es/cmTC_e7ae4.dir/build                                                                                                    
    gmake[1]: Entering directory `/vsc-hard-mounts/leuven-data/333/vsc33366/projects/habitat-sim/build/CMakeFiles/CMakeTmp'
    Building CUDA object CMakeFiles/cmTC_e7ae4.dir/main.cu.o                                                               
    /usr/local/cuda      -c /data/leuven/333/vsc33366/projects/habitat-sim/build/CMakeFiles/CMakeTmp/main.cu -o CMakeFiles/
cmTC_e7ae4.dir/main.cu.o                                                                                                   
    gmake[1]: execvp: /usr/local/cuda: Permission denied                                                                   
    gmake[1]: *** [CMakeFiles/cmTC_e7ae4.dir/main.cu.o] Error 127
    gmake[1]: Leaving directory `/vsc-hard-mounts/leuven-data/333/vsc33366/projects/habitat-sim/build/CMakeFiles/CMakeTmp'
    gmake: *** [cmTC_e7ae4/fast] Error 2
     
     

   

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:12 (project)
Call Stack (most recent call first):
  CMakeLists.txt:12 (project)

-- Configuring incomplete, errors occurred!
See also "/data/leuven/333/vsc33366/projects/habitat-sim/build/CMakeFiles/CMakeOutput.log".
See also "/data/leuven/333/vsc33366/projects/habitat-sim/build/CMakeFiles/CMakeError.log".
Traceback (most recent call last):
  File "setup.py", line 390, in <module>
    include_package_data=True, 
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/site-packages/setuptools/command/install.py", line 67, 
[... 17 lines cut ...]
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/site-packages/setuptools/command/install_lib.py", line 
11, in run
    self.build()
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 204, in run
    self.build_extension(ext)
  File "setup.py", line 273, in build_extension
    env=env,
  File "/data/leuven/333/vsc33366/miniconda3/envs/nav/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/vsc-hard-mounts/leuven-data/333/vsc33366/projects/habitat-sim/src', '
-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/vsc-hard-mounts/leuven-data/333/vsc33366/project
s/habitat-sim/build/lib.linux-x86_64-3.7/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/data/leuven/333/vsc33366/miniconda3/envs/
nav/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=OFF', '-DT
ARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=ON']' returne
d non-zero exit status 1.

My system is CentOS Linux release 7.9.2009 (Core), the nvcc version is 10.0, and python 3.7.  I tried to install the habitat-sim v.0.1.7 version.  Could some kindly inform me how to solve this problem ? Many thanks in advance.

aclegg3, 2022-03-23:
Hey, @Mingxiao-Li 

Looks like an issue with your system cuda install.
You may want to search the repo issue history for similar issues and possible solutions.

@erikwijmans any quick thoughts here?

erikwijmans, 2022-03-23:
Yep, the install is failing when cmake checks to see if it can compile a simple cuda program with the cuda compiler. This indicates an issue with your cuda install.

Mingxiao-Li, 2022-03-24:
> 

Thanks for the reply, but my cuda works fine when I use PyTorch. By the way, is there a way to install the cuda version from Conda ?