facebookresearch/habitat-sim #1717: source installation failure: cmake error

zhixin-zhao, 2022-04-05:
Habitat-Sim version v0.2.1

I followed the source installation instructions in README. Everything goes fine until `python setup.py install` terminal gives such error message:
```bash
CMake Error at cmake/dependencies.cmake:114 (add_subdirectory):
  add_subdirectory given source
  "/home/zhixin/habitat/habitat-sim/src/cmake/../deps/recastnavigation/Recast"
  which is not an existing directory.
Call Stack (most recent call first):
  CMakeLists.txt:113 (include)

CMake Error at cmake/dependencies.cmake:115 (add_subdirectory):
  add_subdirectory given source
  "/home/zhixin/habitat/habitat-sim/src/cmake/../deps/recastnavigation/Detour"
  which is not an existing directory.
Call Stack (most recent call first):
  CMakeLists.txt:113 (include)

CMake Error at cmake/dependencies.cmake:118 (target_compile_definitions):
  Cannot specify compile definitions for target "Detour" which is not built
  by this project.
Call Stack (most recent call first):
  CMakeLists.txt:113 (include)
```

And there is no viewer.py under examples directory. I didn't find any executble py file in the source directory.

zhixin-zhao, 2022-04-05:
I also have tried conda installation(v 0.2.1). Everything work fine but I couldn't find viewer.py file under examples directory either.
Maybe there is a right tutorial for conda habitat which I did't find. Can anyone help me.

dhruvbatra, 2022-04-05:
If conda installation works, I would recommend sticking to it. 

> Everything work fine but I couldn't find viewer.py file under examples directory either.

viewer.py is available from the git cloned habitat-sim directory. Please see this note: 
https://github.com/facebookresearch/habitat-sim/blob/926238c434139dd353615b76b057aad9f69e6a99/README.md?plain=1#L136

closed 2022-04-05 by dhruvbatra