facebookresearch/habitat-sim #1736: Issues with loading URDF

santoz007, 2022-04-28:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
Hi,

I am trying to import a URDF for a robotic system with two manipulators.
I initially tried to import the same system with only one manipulator. It was properly imported. But when it comes to the two manipulator system, I am getting loading issues though the URDF of the two manipulator system was generated in the same way from SOLIDWORKS as the one manipulator system. Any help will be appreciated. 

Note: The way I am generating the URDF is such that each of the two manipulators is a child link of the base link.

aclegg3, 2022-05-02:
Interesting that your URDF isn't working. If there is a single kinematic chain then it sounds like it should be supported. You can turn on [verbose logging](https://aihabitat.org/docs/habitat-sim/logging.html) for the URDF subsystem and try again to see if an obvious error is present. 

If you can share the files here or in a DM on our Slack ([link](https://aihabitat.org/#contact)) I can try them out also.

aclegg3, 2022-05-02:
In this case, the issue was related to a missing `<limit>` element ([SDF reference](http://sdformat.org/spec?ver=1.9&elem=joint#axis_limit) is pretty good) for the joints from the exporter. This element is required by the parser, but not all tags (e.g. "lower" and "upper" are required). If these are not provided, no position limiting will be active by default.

closed 2022-05-02 by aclegg3