facebookresearch/habitat-sim #1741: Can multiple habitat-sim simulators run together from the same thread?

desaixie, 2022-05-03:
I have been using habitat-sim for a while. Recently I am trying to run multiple (currently only 2) habitat-sim simulators simultaneously for multi-scene meta Reinforcement Learning. In my training script, I want to create two simulators on two different scenes, i.e. room_0 and room_1 from Replica, and run my reinforcement learning agent in them at the same time. I create each simulator in a wrapper class 
```
for i, scene_name in scene_list:  
    self.envs.append(WrapperClass(scene_name, ...)
```

and the wrapper class calls `self.sim = habitat_sim.Simulator(cfg)`. 
When I run this, I get SIGSEGV segmentation fault when the second simulator is being created by the `habitat_sim.Simulator(cfg)` call. I tried running my script in gdb, and it shows
```
...
[2022-05-02 23:45:46,003][/home/desaixie/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/logging.py][INFO] - I0502 23:45:46.003570 8627 simulator.py:221] Loaded navmesh ../../../habitat_data/Replica/room_1/habitat/mesh_semantic.navmesh
--Type <RET> for more, q to quit, c to continue without paging--

Thread 1 "python" received signal SIGSEGV, Segmentation fault.
esp::gfx::WindowlessContext::gpuDevice (this=0x0)
    at /opt/conda/envs/py36/conda-bld/habitat-sim_1627787509287/work/src/esp/gfx/WindowlessContext.cpp:116
116	/opt/conda/envs/py36/conda-bld/habitat-sim_1627787509287/work/src/esp/gfx/WindowlessContext.cpp: No such file or directory.
```

However, when I run two training scripts and the same time, each using creating one simulator, there is no error.
Any help is appreciated.

aclegg3, 2022-05-03:
Sorry, you can only have one habitat Simulator object instantiated in a single process. You'll need to use multi-process to run multiple simulators simultaneously.

closed 2022-05-03 by aclegg3

desaixie, 2022-05-03:
Actually, when using habitat-sim installed via building from source instead of conda, I can create two simulators (what I described above) without the above error. Could you fix this for conda installed habitat-sim?