facebookresearch/habitat-sim #1747: importing habitat sim crashes without first importing quaternion

ChanganVR, 2022-05-06:
Branch: sangarg/add-configurable-materials

## 🐛 Bug

Importing habitat sim crashes without first importing quaternion

## Steps to Reproduce

Steps to reproduce the behavior:

1. Build and install with this command: python -u setup.py install --build-type "Release" --lto --audio --headless
2. Open python and import habitat_sim 

## Expected behavior
The program crashes aftering outputing the following info:

free(): invalid pointer
Aborted (core dumped)

## System Info

Platform: Linux-5.4.0-81-generic-x86_64-with-glibc2.17
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.31
Mac version:
Python version: 3.8.12
Architecture: 64bit ELF
Win version:
System OS: Linux
Release: 5.4.0-81-generic
Version: #91-Ubuntu SMP Thu Jul 15 19:09:17 UTC 2021
Operational System: linux
GCC version: b'gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0\n'
CMAKE version: b'cmake version 3.16.3\n'
NVIDIA-SMI: b'Thu May  5 19:16:19 2022       \n+-----------------------------------------------------------------------------+\n| NVIDIA-SMI 470.57.02    Driver Version: 470.57.02    CUDA Version: 11.4     |\n|-------------------------------+----------------------+----------------------+\n| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |\n| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |\n|                               |                      |               MIG M. |\n|===============================+======================+======================|\n|   0  Quadro GP100        On   | 00000000:AF:00.0 Off |                    0 |\n| 26%   33C    P0    30W / 235W |    353MiB / 16278MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n|   1  Quadro GP100        On   | 00000000:D8:00.0 Off |                    0 |\n| 26%   36C    P0    31W / 235W |      4MiB / 16278MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n'
Pip packages versions:
b'imageio-ffmpeg==0.4.5\nnumba==0.55.1\ntqdm==4.62.3\n'
Conda packages versions:
b'attrs                     21.2.0                   pypi_0    pypi  19.1.0\ngitpython                 3.1.24                   pypi_0    pypi\nimageio                   2.9.0                    pypi_0    pypi\nimageio-ffmpeg            0.4.5                    pypi_0    pypi\nmatplotlib                3.4.3                    pypi_0    pypi\nnumba                     0.55.1                   pypi_0    pypi\nnumpy                     1.21.5           py38he7a7128_2  \npillow                    8.4.0                    pypi_0    pypi\nscipy                     1.7.1                    pypi_0    pypi  1.3.0\ntqdm                      4.62.3                   pypi_0    pypi\n

aclegg3, 2022-05-06:
@Skylion007 import fails without importing quaternion before habitat_sim. Thoughts? System package conflicts?

ChanganVR, 2022-07-17:
@Skylion007 is there any update? Other people are also running into this issue. I'd expect more people having the same issue as they start to use the audio library.

Skylion007, 2022-07-17:
Ah, sorry missed this the first time. Hmm, this is odd as our `__init__` package should automatically import quaternion in the proper order to prevent this. @ChanganVR Is this only an issue with the audio module is enabled?