facebookresearch/habitat-sim #1748: Crash while loading mesh.ply from Replica Dataset

RachithP, 2022-05-06:
Habitat-Sim version: v0.2.1
Ubuntu 21.10

## ❓ Questions and Help

I followed [this](https://aihabitat.org/docs/habitat-sim/image-extractor.html) tutorial to extract images from scenes supported by Habitat. 

I'm able to successfully obtain semantic, rgb, depth images from Replica Dataset  scenes under habitat folder, for e.g. `Replica-Dataset/dataset/apartment_0/habitat/mesh_semantic.ply`. However the quality of rgb images here is low.  [Here](https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#replica-dataset) is is suggested to use `mesh.ply` to get better rendering results. I followed the steps to add `sorted_faces.bin` in the respective folders. Now, when I try to load a `mesh.ply` file to extract rgb image, I encounter a crash. I'm not sure what is happening. 

The stdout log is attached here - https://gist.github.com/RachithP/b91212dbfd73a94085e868b6aaf82b9a.

Please let me know if you need more information to address this.

aclegg3, 2022-05-06:
Hey @RachithP, 

Are you loading one scene, or is your script trying to load multiple scenes in sequence? 

The log you attached here indicates that multiple scenes are loaded successfully before the crash of apartment_1. It looks like you are calling `reconfigure()` to switch scenes without calling `close()` and possibly overloading your GPU memory. Replica assets are very large to begin with and PTEX textures further increase their memory requirements.

RachithP, 2022-05-06:
Yes, I'm loading all scenes together as I provided all scenes to the `scene_file_paths` param in `ImageExtractor`
```python
extractor = ImageExtractor(
    scene_file_paths,
    img_size=(256, 256),
    output=["rgba", "semantic"],
    pose_extractor_name="random_pose_extractor"
)
```
I assumed that this would handle the switch between scenes internally as it accepts multiple scenes. But, looks like the memory is overloaded before it can do that. My bad, it works now when I load few scenes at a time.
~

closed 2022-05-06 by RachithP

zhuisa, 2023-07-03:
@RachithP Sorry to bother you, may I ask how to obtain file sorted_faces.bin? The downloaded dataset does not contain file sorted_faces.bin. How can I obtain file sorted_faces.bin.

RachithP, 2023-07-05:
@zhuisa , it looks like the information pertaining to [Replica dataset](https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#replica-dataset) in current main branch is not accurate or incomplete. 

I went back to the habitat version I was working on (v0.2.1) back when I created this issue and found [this](https://github.com/facebookresearch/habitat-sim/blob/v0.2.1/DATASETS.md#replica-dataset). Looks like they had a separate download link to the `sorted_faces.bin` files. Maybe this will be helpful.