facebookresearch/habitat-sim #1750: How to select specific GPUs to for each Habitat-sim instance created in a multi-GPU machine?

desaixie, 2022-05-08:
I am trying to create multiple habitat-sim instances on a node with multiple GPUs using EGL. I have tried setting EGL_DEVICE_ID and CUDA_VISIBLE_DEVICES environment variables. Setting EGL_DEVICE_ID has no effect at all (setting it to 1 yet all simulators are created on GPU 0). Setting CUDA_VISIBLE_DEVICES to 1 works, resulting in all simulations being created on GPU 1 instead of 0. However, when setting CUDA_VISIBLE_DEVICES to 0,1,2, all simulations were still created on GPU 0 even after GPU memory is full. Setting CUDA_VISIBLE_DEVICES to 0 at first and chaning it to 1 after a few simulators are created also did not work, as all simulators are still created on GPU 0.

I saw from other issues they have output like `found 2 EGL devices, choosing EGL device 0 for CUDA device 0`. It seems to come from Habitat-lab, but I am using Habitat-sim only. 

I also saw from [#1038](https://github.com/facebookresearch/habitat-sim/issues/1038#issuecomment-766997942), so my attemps with setting CUDA_VISIBLE_DEVICES mentioned above were done by creating habitat-sim simulators before using CUDA in PyTorch.

erikwijmans, 2022-05-08:
You can set `gpu_device_id` on the config (https://aihabitat.org/docs/habitat-sim/habitat_sim.sim.SimulatorConfiguration.html#gpu_device_id).

> By the way, I am creating all habitat-sim simulators from the same process and the same thread.

This won't work. We don't support multiple simulator instances in the same process/thread.

desaixie, 2022-05-09:
Setting gpu_device_id works for using multiprocessing. However, I previously successfully created multiple habitat-sim simulators in the same process and I wasn't using any multiprocessing/multithreading; this is only for using habitat-sim built from source (it did not work with habitat-sim installed with conda)

erikwijmans, 2022-05-09:
> Setting gpu_device_id works for using multiprocessing

Great! That's the supported way to do what you want.

> However, I previously successfully created multiple habitat-sim simulators in the same process

This isn't something that is supported.

closed 2022-05-09 by erikwijmans