facebookresearch/habitat-sim #1752: generate semantic image in example/viewer.py

Hoyyyaard, 2022-05-13:
command : python examples/viewer.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb

I set the semantic sensor to True but it doesnot show the semantic image

Hoyyyaard, 2022-05-13:
how can I generate the semantic image to the screen?

aclegg3, 2022-05-16:
Hey, @Hoyyyaard. The python viewer is not setup to render semantics or depth images to the canvas at this time. You are welcome to use the C++ viewer for this. 

You can enable a semantic sensor in the python viewer code and use the visualization utilities to save the semantic frames if you prefer to stay in python. You are also welcome to try adding this feature and contributing back it you like.