facebookresearch/habitat-sim #1755: Download Util broken?

uahic, 2022-05-15:
## Habitat-Sim version
 main

## 🐛 Bug
1) In the current conda package, there is no automatic pulling of semantic annotations and this diverts from the description of the install instructions which is quite confusing for beginners

2) I've grabbed the source code from the main branch (utils.dataset_download.py) and tried again. When pulling hm3d_minival,  there is a hm3d-minival-semantic-annots-v0.1.tar.gz file contained but cant be unpackaged.

``` 
Traceback (most recent call last):
  File "download.py", line 688, in <module>
    main(sys.argv[1:])
  File "download.py", line 682, in main
    download_and_place(
  File "download.py", line 520, in download_and_place
    with tarfile.open(data_path + package_name, "r:*") as tar_ref:
  File "/data/miniconda3/lib/python3.8/tarfile.py", line 1608, in open
    raise ReadError("file could not be opened successfully")
tarfile.ReadError: file could not be opened successfully
```

The file is 324 bytes large and can't even be unpacked using tar 

``` 
tar -zxvf hm3d-minival-semantic-annots-v0.1.tar.gz
tar (child): hm3d-minival-semantic-annots-v0.1.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
 ```

Is the file corrupt or is it another format than the file extension suggests?

joaomcm, 2022-05-18:
I'm having the same issue when attempting to download the semantic labels for the validation and train splits. Did you find any solution to this?

StOnEGiggity, 2022-05-25:
Hi, I have the same issue, too. It seems that the progress is stuck when it tries to unpack "hm3d-minival-habitat.tar", and I have applied for the access to HM3D dataset. Is there any suggestion?

dhruvbatra, 2022-06-03:
@uahic @joaomcm -- can you try again now?

uahic, 2022-06-06:
@dhruvbatra works for me now. It seems that only very few scene folders contain dedicated semantic annotation files, probably it is what it is but maybe it could be another bug

Thanks for the fix

uahic, 2022-06-06:
@dhruvbatra works for me now. It seems that only very few scene folders contain dedicated semantic annotation files, probably it is what it is but maybe it could be another bug. 

Thanks for the fix

joaomcm, 2022-06-08:
@dhruvbatra Thank you! 
The util is now fully working. And the semantic annotations for the training set are now being fully downloaded (before, only 10 of the scenes had their semantic .glb loaded).
One interesting thing to note is that it seems that the class labels in natural language are somewhat consistent (even though in some scenes a bed covered in a blanket is labelled as a blanket), the class ID numbers are not consistent between classes ,i.e. , a semantic_id= 30 has a category.name = 'chair' and a category.index = 1 in a scene, but on a different scene, another instance with semantic id (for instance) 45 has the category name "chair', but the category.index = 15 - i.e. , the mapping between category_names and category_indices is not consistent between scenes. Is that by design of is there a bug in the labelling?

Kind regards,

Joao

dhruvbatra, 2022-06-08:
CC: @srama2512

srama2512, 2022-06-09:
@joaomcm  - What class ID numbers are you referring to here? Can you please share a script to reproduce the above issue?

srama2512, 2022-06-27:
@joaomcm  - Apologies for the latency. The issue stems from the fact that the object names are not consistently mapped to the semantic categories in habitat-sim. There are no plans to currently address this. If you'd like to take a stab at it, @aclegg3 may have some suggestions. 

At the moment, I map the raw text name to the matterport categories pythonically based on [this TSV](https://github.com/facebookresearch/habitat-sim/blob/main/data/matterport_semantics/matterport_category_mappings.tsv). That is, map from `raw_category` / `category` columns to `mpcat40` column. You can then map it to the integer index of interest.

joaomcm, 2022-08-04:
Thank you, @srama2512 - I'll proceed with mapping the classes using their names, then.

Once again, thank you for getting back to me.

aclegg3, 2022-08-04:
@joaomcm,

I agree it would be useful to support the mapping of free form instance annotations back to a standard set of categories as @srama2512 linked above.

I'm sure part of this will require some standardization in the free form entry classes followed by a programmatic utility. Feel free to consider contributing such a tool and documenting any issues you find in the text files.

closed 2022-08-18 by aclegg3

joaomcm, 2022-08-19:
Dear @aclegg3,

Sure thing. If I end up doing that on my end I'll submit a PR with the correction tool and documenting the labeling differences. Hopefully It'll be useful to the community!

Kind regards,

Joao.