facebookresearch/habitat-sim #1765: Build from source --with-cuda, ImportError:_magnum.cpython-36m-x86_64-linux-gnu.so: undefined symbol

Zhongxyyyy, 2022-05-26:
My environment:
-     Nvidia Driver Version: 470.103.01
-     CUDA Version: 11.4
-     Operating System + Version: Ubuntu 20.04
-     Python Version: 3.6.13
 
I tried to build habitat-sim from source with cuda, 

```
git clone --branch v0.1.7 git@github.com:facebookresearch/habitat-sim.git
cd habitat-sim
sudo apt-get update || true
sudo apt-get install -y --no-install-recommends \
     libjpeg-dev libglm-dev libgl1-mesa-glx libegl1-mesa-dev mesa-utils xorg-dev freeglut3-dev
python setup.py install --with-cuda
```
And then, I got 
```
....
Finished processing dependencies for habitat-sim==0.1.7
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Processing ./build/deps/magnum-bindings/src/python
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: magnum
  Building wheel for magnum (setup.py) ... done
  Created wheel for magnum: filename=magnum-0.0.0-cp36-cp36m-linux_x86_64.whl size=55937946 sha256=6cd97100f893559a336689e28be2c7d76165c723407359409f9ee542e0d4f310
  Stored in directory: /tmp/pip-ephem-wheel-cache-qxnqfw0w/wheels/eb/87/f7/082b3de175d7c19286f2f020578c0e80c4aa6379af7c2c02ed
Successfully built magnum
Installing collected packages: magnum
  Attempting uninstall: magnum
    Found existing installation: magnum 0.0.0
    Uninstalling magnum-0.0.0:
      Successfully uninstalled magnum-0.0.0
Successfully installed magnum-0.0.0
```
It seems successful, but when I run the demo code of VLN-CE,  I got an error:
```
Traceback (most recent call last):
  File "/home/kk/.conda/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.7-py3.6-linux-x86_64.egg/habitat_sim/agent/agent.py", line 10, in <module>
    import magnum as mn
  File "/home/kk/.conda/envs/habitat/lib/python3.6/site-packages/magnum/__init__.py", line 28, in <module>
    from _magnum import *
ImportError: /home/kk/.conda/envs/habitat/lib/python3.6/site-packages/_magnum.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6Magnum2GL7Context13ConfigurationC2Ev

Process finished with exit code 1
```
I think it may related to the magnum version (before is 0.0.0) , I tried to reinstall the magnum with version 14.0.0
```
pip uninstall magnum 
pip install magnum 
```
but I got another error 
```
ModuleNotFoundError: No module named 'magnum.scenegraph' 
```
Please help

erikwijmans, 2022-05-26:
Please let habitat's install manage magnum to make sure the magnum install is compatible, don't install it directly from pip.

Have you installed habitat-sim via conda before in this environment? If so, either uninstall it before building from source or create a new env as conda install and building from source in the same env can cause weird errors like this.

Zhongxyyyy, 2022-05-28:
Thanks your prompt reply! I know magnum should compatible with habitat version, but when I build habitat-sim from source, it install magnum version 0.0.0 automatically, is it correct?
Yes, I tried to conda install and build from source in this environment, maybe it's the problem, I create a new env and try again.

erikwijmans, 2022-05-29:
Yep, 0.0.0 is correct

aclegg3, 2022-08-30:
Assuming this issue is resolved and closing.

closed 2022-08-30 by aclegg3