facebookresearch/habitat-sim #1768: Files for Franka robot

arjung128, 2022-05-31:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

I am trying to use this with the Franka robot, but I would need something analogous to `robots/fetch_robot.py` for the franka robot... where can this be found? Thanks!

aclegg3, 2022-05-31:
Hey @arjung128,

There is not yet any Habitat metadata for the Franka at this time. That said, Habitat-sim will support most URDF files if you can find them. (For example, a quick search turns up [options like this](https://github.com/StanfordASL/PandaRobot.jl/tree/master/deps/Panda)). With this alone, you can load the robot into Habitat, control joints, create and actuate motors, etc... 

The fetch_robot.py primarily defines parameters for the mobile_manipulator.py to make control a bit easier and abstract away the details. It isn't necessary, but you may find that using it is a bit easier after you've taken the time to set it up. You could define the same parameters for your robot in a new file. 

Since the Franka is not a mobile platform you may choose to implement an intermediate manipulator.py class or ignore/override the mobile base components.

arjung128, 2022-05-31:
Got it, thanks!

aclegg3, 2022-08-18:
Robotics wrappers and support have been migrated to habitat-lab.

closed 2022-08-18 by aclegg3