facebookresearch/habitat-sim #1769: Incorrect positions for Replica Dataset?

cwchenwang, 2022-05-31:
## Habitat-Sim version
v0.2.1

## ❓ Questions and Help
Not really sure this is a habitat question nor replica question. I first use habitat to random select valid indexes and print out their positions. Then I input these positions to the replica renderer and renders a six-image cubemap, however the cubemap looks pretty weird, shows below:

[image]

Is this because the positions in habitat cannot be directly interpreted as the position in the world coordinate?

erikwijmans, 2022-05-31:
Please follow https://github.com/facebookresearch/Replica-Dataset/issues/29 to convert between the coordinate frames.

closed 2022-05-31 by erikwijmans

cwchenwang, 2022-06-01:
@erikwijmans Thanks for your help. I seemed to move my replica camera to the right spot. However, there is still a problem: the position of my camera is very low. You can see from the two images I uploaded here: the upper one is from habitat sim and the lower one is from ReplicaSDK. Both of them stand on the same location, but the camera in the ReplicaSDK seems to be put on a very low height. 
[image]
[image]

Some details: I used the `apartment_0` scene, and the topdown coordinate of the position is (40, 54), the position in habitat is (2.1072494029998783, -1.374765157699585, -0.9503817558288574), which is converted to (2.1072495, 0.95038176, -1.3747652) with the pointed script. And I input the converted position to the ReplicaSDK, here is my replica code:

std::vector<float> position = {2.1072495, 0.95038176, -1.3747652, };

  int cx = rand()%4;
  int cy = rand()%4;

  // Setup a camera
  pangolin::OpenGlRenderState s_cam(
      pangolin::ProjectionMatrixRDF_BottomLeft(
          width, height,
          width / 2.0f, width / 2.0f, // focal length
          (width - 1.0f) / 2.0f, (height - 1.0f) / 2.0f, // camera center
          0.1f, 100.0f), // near and far
      pangolin::ModelViewLookAtRDF(position[0], position[1], position[2], 
                                   cx, cy, position[2], 
                                   0, 0, 1));

erikwijmans, 2022-06-11:
Habitat talks in agent position while Replica talks in camera position. The default camera height in habitat is 1.5m, so add 1.5m to the y component before converting.