facebookresearch/habitat-sim #1770: ArmEEAction does not support grasping?

arjung128, 2022-05-31:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Is there any reason why 'ArmEEAction' does not support grasping, unlike `ArmAction`?

aclegg3, 2022-06-13:
@ASzot may have more context here.

aclegg3, 2022-08-18:
All robotics wrappers and grasping support is now in habitat-lab.

closed 2022-08-18 by aclegg3