facebookresearch/habitat-sim #1773: Get the amodal semantic mask (containing occlusions) of a target object in the scene

LeiFan95, 2022-06-01:
Hi, I am trying to get the amodal semantic mask of any object in the scene. I wonder if there is any feature to do this? 

I checked the previous pull [#668](https://github.com/facebookresearch/habitat-sim/pull/668) on configurable semantic ID, but it is about adding new objects. However, I hope to get the mask of existing objects.

Any help would be appreciated.

aclegg3, 2022-06-13:
Hey @LeiFan95,

Am I correct that what you'd to produce is an object mask image where pixels containing the target object are one color and all other pixels are black?

If so, I don't think there is any functionality to explicitly do this, but there are ways to post-process sensor data to produce such an image. For example, if you know the desired instance id for the target object, you could generate this mask through an image transform from the instance-level semantic sensor.

aclegg3, 2022-08-30:
Haven't heard anything back here, assuming resolved and closing issue.

closed 2022-08-30 by aclegg3

KevinZ-01, 2023-12-12:
@LeiFan95 Hi Lei, I am facing the same problem here. Have you solved this? 
@aclegg3 The instance map from the instance-level semantic sensor does not include the occluded part of an object. Is there a way to get the mask (including the occluded part) of an existing object in an environment? Thanks for any help!

LeiFan95, 2023-12-13:
@KevinZ-01 Hi Kevin, I don't think there is a feasible way in Habitat to directly obtain an unoccluded mask of any target. However, if you need to get unoccluded masks for a small number of objects from different viewpoints, you could try modifying the 3D scene model to be loaded. For example, you could remove all meshes in the 3D scene model except for the object of interest. Thus, when projecting a 2D image from this modified 3D scene model, you would be able to obtain an intact mask of the target without any occlusions. The whole process takes a lot of work and is quite time-consuming. 
Or, maybe you can change the projection code to only project the object of interest when getting 2D masks. For example, you can set other objects to be transparent when doing projection. I haven't tried this way before. It is just a thought. 😀

KevinZ-01, 2023-12-24:
> @KevinZ-01 Hi Kevin, I don't think there is a feasible way in Habitat to directly obtain an unoccluded mask of any target. However, if you need to get unoccluded masks for a small number of objects [...]

Thank you! It helps a lot :)