facebookresearch/habitat-sim #1782: How to load replica dataset?

20481535, 2022-06-13:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

20481535, 2022-06-13:
I see all the tutorials use glb file to load.
However, the downloaded datset are ply file. How to load it?

aclegg3, 2022-06-14:
Check out the home page of the [Replica-Dataset](https://github.com/facebookresearch/Replica-Dataset) for instructions on how to test the dataset within Habitat. 

Essentially you'll just want to run the newest version of the Replica downloader script and then set `SimulatorConfiguration.`[scene_dataset_config_file](https://aihabitat.org/docs/habitat-sim/habitat_sim.sim.SimulatorConfiguration.html#scene_dataset_config_file)` = /path/to/Replica/replica.scene_dataset_config.json` and `SimulatorConfiguration.`[scene_id](https://aihabitat.org/docs/habitat-sim/habitat_sim.sim.SimulatorConfiguration.html#scene_id) to the name of the scene (.e.g. `frl_apartment_0`).

closed 2022-06-14 by aclegg3