facebookresearch/habitat-sim #1791: Installing 'habitat_sim.sensors.noise_models.data' as data is deprecated, please list it in `packages`.     !!

gudehhh666, 2022-06-23:
Hello all;
I was getting following error while running : python setup.py install

Installing 'habitat_sim.sensors.noise_models.data' as data is deprecated, please list it in `packages`.
    !!

    ############################
    # Package would be ignored #
    ############################
    Python recognizes 'habitat_sim.sensors.noise_models.data' as an importable package,
    but it is not listed in the `packages` configuration of setuptools.

    'habitat_sim.sensors.noise_models.data' has been automatically added to the distribution only
    because it may contain data files, but this behavior is likely to change
    in future versions of setuptools (and therefore is considered deprecated).

    Please make sure that 'habitat_sim.sensors.noise_models.data' is included as a package by using
    the `packages` configuration field or the proper discovery methods
    (for example by using `find_namespace_packages(...)`/`find_namespace:`
    instead of `find_packages(...)`/`find:`).

    You can read more about "package discovery" and "data files" on setuptools
    documentation page.

And I totally couldn't handle it.
Can anyone please help me

erikwijmans, 2022-06-23:
What python version is this? We haven't seen this.

gudehhh666, 2022-06-23:
python 3.8.12
It's so strange, I wonder how to handle this.

aclegg3, 2022-08-30:
Possibly a unique system issue, as we haven't seen this. Suggest you try again with a fresh conda env. 
We'd need more information (e.g. system, install process, logs, etc...) to offer any advice if the issue persists. 
@gudehhh666 we'll close this as stale if we don't hear from you in the next week or two.

aclegg3, 2022-11-02:
Closing as stale and unverified.

closed 2022-11-02 by aclegg3