facebookresearch/habitat-sim #1794: How to get the bounding box of the scene ?

derrick-xwp, 2022-06-28:
## ❓ Questions and Help
How to get the bounding box of the scene ?
# In world coordinate?

yaelbenoren, 2022-07-03:
bounds = sim.pathfinder.get_bounds()

aclegg3, 2022-08-18:
You can also compute the cumulative AABB for the SceneGraph:

scene_graph = sim.[get_active_scene_graph()](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#get_active_scene_graph)
root = scene_graph.[get_root_node](https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SceneGraph.html#get_root_node-6eef6)()
root.[compute_cumulative_bb](https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SceneNode.html#compute_cumulative_bb)()
scene_bounding_box = root.[cumulative_bb](https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SceneNode.html#cumulative_bb)

closed 2022-08-18 by aclegg3