facebookresearch/habitat-sim #1801: Problem with calculating large navmeshes

yaelbenoren, 2022-07-03:
I am trying to run habitat with my own downloaded scenes. However, It seems that if a scene is large (about 200mX200m) it is not possible to calculate a correct navmesh for it and I get this error:
RuntimeError: NavMesh has no navigable area, this indicates an issue with the NavMesh
Is there a way to fix it?
Thank you

erikwijmans, 2022-07-04:
Please provide the error that occurs when building that navmesh.

yaelbenoren, 2022-07-06:
CREATING NAVMESH...
I0706 16:48:58.650434 31949 PathFinder.cpp:382] Building navmesh with 4997x4431 cells
E0706 16:49:07.447821 31949 PathFinder.cpp:624] Could not build Detour navmesh
E0706 16:49:07.481483 31949 Simulator.cpp:786] Failed to build navmesh
RuntimeError: NavMesh has no navigable area, this indicates an issue with the NavMesh

Skylion007, 2022-07-13:
@yaelbenoren You may also want to try to increase `detailSampleMaxError` as that is one of the params of the step that is failing.

closed 2022-08-18 by aclegg3