facebookresearch/habitat-sim #1805: Color topdown map

jingjing00kk, 2022-07-08:
Hello,
I have a question about top down map. We can receive a black and white top down map using function "get_topdown_view".
Can I get color topdown map like this? And how?
[image]
Thank you!

closed 2022-07-08 by aclegg3

jingjing00kk, 2022-07-11:
Thank you @aclegg3 

But when I edit the hm3d .glb file with Blender, there are only meshes. Textures are not shown in the Blender (the mode of Blender is changed to texture mode, and I have test other .glb files, and the textures were displayed well.) . Also I tried 3dmax, micrsoft 3D viewer, the .glb files could not be opened by these softwares. Are the .glb formats  in the HM3d datasets specialized or different with normal .glb files or the version of the software should be a some special one?

aclegg3, 2022-07-11:
@jingjing00kk, there are multiple downloads for HM3D. The habitat download is basis compressed for more efficient use in simulation. Blender won't load those. The other formats should contain uncompressed textures for the same scenes which can be loaded in Blender.