facebookresearch/habitat-sim #1813: Building with --audio causes

drydenwiebe, 2022-07-15:
## Habitat-Sim version
v0.2.2

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Yes.

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Yes.

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

System: 
Ubuntu 21.10 x86

This is half a question and half a bug report.

How I got this error:

```
conda create -n habitat-sim python=3.7 cmake=3.14.0 -y 
conda activate habitat-sim
pip install -r requirements.txt
git clone --branch v0.2.2 https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
python setup.py install --bullet --audio
python -m habitat_sim.utils.datasets_download --uids replica_cad_dataset
```

Causes:

```
free(): invalid pointer
Aborted (core dumped)
```
This same issue happens when using habitat-sim in sound-spaces (although that is out of scope for this question).

However running examples such as ```examples/viewer.py``` works the same as when it is compiled without --audio. I was wondering if anyone had ran into this before when trying to build v0.2.2 with new audio support.

Thank you!

dhruvbatra, 2022-07-16:
@ChanganVR

ChanganVR, 2022-07-17:
Hi @drydenwiebe this is an existing and known bug (see the open issue here: https://github.com/facebookresearch/habitat-sim/issues/1747) and I'm also waiting for other people to troubleshoot this.

When this error happened, importing quaternion or magnum before habitat_sim somehow magically worked for me earlier. You could give it a try.