facebookresearch/habitat-sim #1814: Questions about use replicaCAD

yaofeng11, 2022-07-16:
## Habitat-Sim version
v0.2.2
Hi there!
I am using MP3D datasets and my code is 
` test_scene = "home/yaofeng/PycharmProjects/habitat/testscene/scene_datasets/hm3d-val-habitat/00885-dVW2D7TDctW/dVW2D7TDctW.basis.glb"
agent_num = 2
rgb_sensor = True  

sim_settings = {
    "width": 256,  # Spatial resolution of the observations
    "height": 256,
    "scene": test_scene,  # Scene path
    "default_agent": 0,
    "sensor_height": 1.5,  # Height of sensors in meters
    "color_sensor": rgb_sensor,  # RGB sensor
    "depth_sensor": depth_sensor,  # Depth sensor
    "semantic_sensor": semantic_sensor,  # Semantic sensor
    "seed": 1,  # used in the random navigation
    "enable_physics": False,  # kinematics only
}`
and I found that the frame is not very clear
[image]
So I decided to use replicaCAD  and I also found someone asked the same question about how to use this dataset
I mean that if I use this replicaCAD all I need to change is to change the test_scene like
test_scene = "/home/yaofeng/PycharmProjects/habitat/data/replica_cad/replicaCAD.scene_dataset_config.json --scene apt_1"
Because I have already use MP3D for  testing mulitiagent cooperation which I mean the way you shown on colab is common to replicaCAD datasets.

And another question for python viewer.
I test replicaCAD like you shown
`python examples/viewer.py --dataset /home/yaofeng/PycharmProjects/habitat/replica_cad/replicaCAD.scene_dataset_config.json --scene apt_1`
and it rise 
`Traceback (most recent call last):
  File "examples/viewer.py", line 27, in <module>
    class HabitatSimInteractiveViewer(Application):
  File "examples/viewer.py", line 719, in HabitatSimInteractiveViewer
    def exit_event(self, event: Application.ExitEvent):
  AttributeError: type object '_magnum.platform.glfw.Application' has no attribute 'ExitEvent'
`
I don't know whats' wrong? Can someone help me.Thanks!

aclegg3, 2022-08-01:
Hey, @yaofeng11 

> I found that the frame is not very clear

You are specifying a 256x256 image resolution for the render here. If you want a clearer image, you should use a higher resolution.

aclegg3, 2022-08-01:
>I test replicaCAD like you shown
python examples/viewer.py --dataset /home/yaofeng/PycharmProjects/habitat/replica_cad/replicaCAD.scene_dataset_config.json --scene apt_1

This works for me locally, so maybe you're on an old version or have some system specific issue? 
Does the C++ viewer (`./build/viewer`) work for you? 
Have you tried re-installing `main` branch from source in a fresh conda environment, downloading the most recent ReplicaCAD, and then running the viewer?

closed 2022-08-12 by aclegg3